ChefKissInc/NootedRed #3: Some more info about the project

sdassh, 2022-08-01:
Is this kext daily usable? Does everything work? Is hardware acceleration working? Could you put some more info in readme?

ChefKissInc, 2022-08-01:
@sdassh No to all, it’s in active research/development, hence why all the debug commits.

handicraftsman, 2022-08-12:
Some more questions:

Do you have any progress tracker?

When prebuilt binaries are going to become available? (I am asking for the condition, not ETA)

Are there going to be nightly builds, e.g. for testing?

ChefKissInc, 2022-08-12:
@handicraftsman No progress tracker, live progress is done in the telegram group of @mvaisakh, see https://github.com/mvaisakh/Asus-X505ZA-Ryzen-Hackintosh/issues/1
A release will be made once all the AMD kexts succeed initialisation and startup, and macOS boots to the desktop on my Picasso test laptop
I don’t have CI set up because it seems useless for now, as any testing of these kexts won’t yield any different result than on our side, and people might get confused and start making issues about how the builds don’t work

commit 4a1677ce (2022-08-01): Add information in the README

Closes #3
You're welcome.
closed 2023-02-25 by ChefKissInc