ChefKissInc/NootedRed #10: Fix display

ChefKissInc, 2023-02-22:
All displays are blank. Investigation is in progress. This is for progress tracking

someusern, 2023-02-22:
look whatever green black display fix (maybe this is the only solution?)

SsNiPeR1, 2023-02-22:
@someusern this kext does not work without whatevergreen at all bro

ChefKissInc, 2023-02-22:
@SsNiPeR1 Please do not respond on our behalf! And this seems to be incorrect information

ChefKissInc, 2023-02-22:
@someusern The display is not being programmed correctly as it is newer technology than the GCN 5 (AMDRadeonX5000) kexts expect. This is not a simple “duct tape”-able issue. 
Also, we’re already using WhateverGreen, but we aren’t sure how much patching it’s doing as there is a Lilu bug that prevents multiply-nested devices from being detected (such us AMD iGPUs, path for 99% of dGPUs is something like PCI0.GPP8.blah, while for AMD iGPUs it is PCI0.GPP8.IOPP.blah)
We wish it was as simple as just some random black screen fix. Haha

someusern, 2023-02-22:
I understand

ChefKissInc, 2023-02-23:
Resolved in latest commits!

closed 2023-02-23 by ChefKissInc