ChefKissInc/NootedRed #11: Fix Renoir support

ChefKissInc, 2023-02-22:
Renoir does not work at all. Reason unknown. Will be investigated after the display issue is resolved. This is for progress tracking

dhr-uvin, 2023-02-24:
Hello.

I am a person whos been following this project for a while , mainly for renoir. The README doesnt mention if it will be fixed , so are attempts to fix it happening?

Thanks

ChefKissInc, 2023-02-24:
@dhr-uvin If there was no intention to fix it, why would we mention it, even make an issue about it?
We’re not sure what the problem is, so fixing it has got a long way. The error is too vague to do anything with

dhr-uvin, 2023-02-25:
> make

Sorry , just asking. Keep up the good work!

akwangl, 2023-03-08:
Look forward to 5000 Series

dockylf, 2023-03-08:
Looking forward to 6900hx

ChefKissInc, 2023-04-21:
@jalavoui Interesting idea. We’ll try that on our next testing rounds, see where that leads

ChefKissInc, 2023-04-30:
@jalavoui Seems to be the Accelerator specifically; it dies randomly while processing some SDMA commands that WindowServer sends early in initialisation. FB-only mode works, and the HWLibs themselves are unlikely to be causing the issue on their own

danboeira, 2023-05-01:
Hello everyone, first I would like to thank you for your commitment to the development of this kext.
I would like to make myself available as a tester for the AMD Ryzen 5 4600G processor, with the Vega 7 (Ranoir) on my computer.
At the moment I will not be able to invest in a compatible GPU, and I would like to test this new kext on my computer.
So far, I haven't been able to boot as needed with the guidelines through the available ''read me''.
Thank you very much!!!

hsnfirdaus, 2023-05-04:
I think this also affected Ryzen 5 5600G. I hope you can find a solution (for now using -nredfbonly boot flag "Partialy accelerated" work)!

ChefKissInc, 2023-05-04:
@hsnfirdaus The title says 4XXX+…

Berkan111111, 2023-05-08:
Will run amd ryzen 5 5500u processor

akwangl, 2023-05-08:
> Will run amd ryzen 5 5500u processor

Don't dream，We can't do that yet

PsaTyrE, 2023-05-09:
> I think this also affected Ryzen 5 5600G. I hope you can find a solution (for now using -nredfbonly boot flag "Partialy accelerated" work)!

can i ask for your efi file

commit 0794e89e (2023-05-21): Fix Renoir

Fixes #11
closed 2023-05-21 by ChefKissInc