ChefKissInc/NootedRed #13: Safari/WebKit leads to constant WindowServer crashes and eventually macOS crash/hang

ChefKissInc, 2023-02-22:
Seems to be causing GPU page faults. Related dmesg attached below
[dmesg-1676548999.txt](https://github.com/NootInc/WhateverRed/files/10807078/dmesg-1676548999.txt)

ChefKissInc, 2023-02-23:
Issue severity has been reduced. Only some videos cause this now

ChefKissInc, 2023-03-01:
Seems to be related to VRAM size. Increasing VRAM on my laptop from 512MB to 1GB resolves it. Maybe because macOS thinks it's a dGPU, it's not optimising how much VRAM it can use. Requires more research

ChefKissInc, 2023-03-31:
Hey @jalavoui. I've been taking peaks at your thread. You've given us a few useful clues here and there for a few of these fixes. Thanks for your suggestions. Would you mind explaining what this patch does?
P.S. It would be nice if you joined our Telegram group. It is here: https://t.me/NootInc

bobouncle, 2023-06-12:
todesk application cannot be used.  is crash

hoppingninja666, 2023-06-17:
Playing a YouTube video Firefox causes hangup of the whole system and does not recover from this state. I am using macOS Monterey. In chrome and chrome based browsers, first few minutes when you use the search bar, the whole browser window turns pinkish and recovers after a while, and becomes smooth after a few of such hangup. I could play in 4k using chromium based browsers.

ChefKissInc, 2023-06-17:
@hoppingninja666 It’s weird, some systems have problems with Chrome, some with Firefox, some not at all. Pretty weird

ChefKissInc, 2023-06-18:
macOS wasn't brave enough to handle it

war1644, 2023-07-03:
me too
1. Continuous long stuttering when entering text
2. There is a lag when opening the image
3. Other

war1644, 2023-07-03:
> 现在我用最新的驱动文件，也遇到了同样的情况，但是不会崩溃，一张一张卡的图片，整个画面卡了一会儿，不动了，过一会儿就可以用了

+1

ChefKissInc, 2023-07-03:
I have not forgotten about the issues; you don't need to create bumps.

sinsin700, 2023-07-05:
I can walk with Brave Browser but lag when opened I try to wait for it to be opened

joederpoliveira, 2023-07-06:
First... Sorry for my bad english.

But. If you need Chrome like I do... This is what I did:

1) Install Chrome (duhh);
2) Open Terminal and paste "/Applications/Google\ Chrome.app/Contents/MacOS/Google\ Chrome --args --disable-gpu" (without quotes);
3) Go to Chrome's settings and look for something like "Use hardware acceleration" and disable this option;
4) restart the browser.

I've done that and still haven't had any issues.

ChefKissInc, 2023-07-06:
@joederpoliveira https://nootinc.github.io/nred#chrome-chromium-based-browsers-and-apps-like-sublime-text-cause-graphical-artefacts-amongst-other-problems

joederpoliveira, 2023-07-06:
> @joederpoliveira https://nootinc.github.io/nred#chrome-chromium-based-browsers-and-apps-like-sublime-text-cause-graphical-artefacts-amongst-other-problems

The problem is that opening Chrome already caused the crash. I just got it the way I posted above.

joederpoliveira, 2023-07-06:
I didn't even have time to enter chrome://flags.

ChefKissInc, 2023-07-06:
@joederpoliveira Sorry, I just meant that I already have this as an entry

joederpoliveira, 2023-07-06:
> @joederpoliveira Sorry, I just meant that I already have this as an entry

No problem. It turns out that my post can help someone who is not able to get into Chrome to disable the flag. That's why I decided to post it here. I think it should work on other Chromium based browsers.

gorouflex, 2023-07-16:
If I use Microsoft Edge and watch a video on that, it will crash (or blink the screen) for some reasons, also that may happen to some application. Can be fixed by turn off **`Hardware Acceleration`** setting.

winktool, 2023-08-19:
> Seems to be related to VRAM size. Increasing VRAM on my laptop from 512MB to 1GB resolves it. Maybe because macOS thinks it's a dGPU, it's not optimising how much VRAM it can use. Requires more research

plz how to change vram?

ChefKissInc, 2023-08-19:
@winktool Smokeless-UMAF

winktool, 2023-08-19:
> Smokeless-UMAF

love you

Woohwah, 2023-08-23:
On Ventura I got this issue when I open the App Store

liuliu11ss, 2023-08-27:
5600H CPU opens an application or a page of the operating system will get stuck, and the system will respond later. The latest version of the driver, MAC OS is ventura.

VisualEhrmanntraut, 2023-08-27:
Thanks, very useful

VisualEhrmanntraut, 2023-09-04:
I am working on a potential fix. I think what I'm implementing will completely fix this and #137. No promises though.

commit fb1bb3dd (2023-09-06): Inject all correct firmware. System is now smooth and cool.

Who said macOS on AMD laptops and couldn't work well?
Fixes #137
Fixes #13
Tested-by: ExtremeXT
Tested-by: lev666
Tested-by: Rxxbertx
Tested-by: Yan Schafer
Tested-by: Mert C
Tested-by: Pablom
Tested-by: Truong Tuan
Tested-by: Sapphire
Tested-by: iVintagelover
closed 2023-09-06 by VisualEhrmanntraut