ChefKissInc/NootedRed #14: ReadME- define specifically what is iGPU-only hackintoshes

ketonik, 2023-02-24:
Hi, readme says “ iGPU-only hackintoshes, specifically Raven/Raven2/Renoir and their derivatives, such as Picasso.”

Can you define exactly what Ryzen CPU’s have the above iGPU families?
Example for the Ryzen 9 7950X, the only iGPU details provided on AMD product website is ‘AMD Radeon™ Graphics’. 

Cannot find any reference what ‘AMD Radeon™ Graphics’ is on this CPU. Certainly no reference to Raven/Raven2/Renoir and their derivatives, such as Picasso’…

Might help others too wanting this CPU or 7950X3D on an ITX board, using internal graphics for display so as to use the only PCIe slot for a Audio Interface PCIe card direct to the CPU…graphics and gaming have zero priority or interest here.

Allen050329, 2023-02-24:
The "iGPUs" mentioned here are ones based on the Vega architecture, so basically only laptop CPUs/APUs that are based on Zen/Zen+/Zen2/Zen3 (not including Zen3+ ), aka Ryzen 2000~5000 with iGPUs (no Ryzen 6000 and beyond) will be able to leverage this kext

ketonik, 2023-02-24:
Thx for the clarifications because now I finally know iGPU on any current Intel or AMD CPU is impossible, now and forever (so forced into the Mac Mini/Thunderbolt solution…so expensive$$$). WhateverRed will be important for other use cases, all the best on that. Thx

chrismin13, 2023-02-24:
> The "iGPUs" mentioned here are ones based on the Vega architecture, so basically only laptop CPUs & desktop APUs that are based on the Zen/Zen+/Zen2/Zen3 CPU architectures (not including Zen3+), aka Ryzen 2000~5000 with iGPUs (not including Ryzen 6000 and beyond) will be able to leverage this kext; sorry for the confusion.
> As for how to know the iGPU family, I'd suggest you turn to, in no particular order, Aida64, HWInfo64, GPU-Z or similar tools.

Readme says that there is no support for Renoir, so that excludes Ryzen 4000 and 5000, or am I missing something? I'd certainly be willing to try it on a Ryzen 5300U if that's not the case.

ChefKissInc, 2023-02-24:
@chrismin13 No, at the moment anything Renoir (like 4000 and 5000 series) does not work. Resolution of this problem is next on our backlog

commit cac8112c (2023-02-24): Rewrite README

Closes #14
closed 2023-02-24 by ChefKissInc