ChefKissInc/NootedRed #20: Amd Ryzen 3200u, vega3 stuck

FlyingThaCat, 2023-03-08:
Im already use the latest lilu and I download from the latest CI and when i use it, it stuck on this progress what should I do
[image]

ChefKissInc, 2023-03-08:
These issues aren't for requesting support. Use of experimental software means you will have issues. Some common user-side configuration problems are: Not using MacBookPro16,3 or MacPro7,1 SMBIOS, using WhateverGreen, not using Lilu 1.6.4 or later (Official release was made last Monday). Some issues we don't know how to resolve yet are: random AGDP-related black screens, and sometimes, but very rarely, random early panics

ChefKissInc, 2023-03-08:
Oh, and, recoveryOS might get stuck in verbose using this kext (including updates), there is an attempt made to fix it, but its effectiveness is not apparent yet

closed 2023-03-08 by ChefKissInc