ChefKissInc/NootedRed #28: DRM not functional

ChefKissInc, 2023-03-17:
```
GVA error: AVD_api.Create failed err=5
GVA error: flow_KE1 physicalAcceleratorCreate 7
GVA error: AVD_api.Create failed err=5
GVA error: Flow_QT0::physicalAcceleratorDispose, m_physical_accel_created == false
GVA error: AVD_api.Create failed err=5
GVA error: Flow_QT0::physicalAcceleratorDispose, m_physical_accel_created == false
```

ChefKissInc, 2023-05-05:
@jalavoui Well the DRM-unlocking code is just UnfairGVA from WhateverGreen. So if it doesn’t work, it’s likely their fault

ezxpro, 2023-05-05:
Is this DRM-related black screen mentioned by @jalavoui something that only happens when playing DRMed content or is there aa possibility this good happen at boot too?

I sometimes get a black screen at boot, when that happens nothing seems to fix it, I have to remove nooted red and reinstall Big Sur again. It works fine until I get this black screen, then it will never boot with nooted red again.
Curiously this also happened to some fresh installs I made in the last several days. It boots fine once or two times, then once I get the black screen and it no longer boots.

Kanepan, 2023-08-25:
Can this be fixed? Looking forward to solving this problem, thank you
Sorry, I posted it again

aleelmaitro, 2023-08-26:
Hello @ChefKissInc ...
I have made a successful publication with a Ryzen 5 3400G, and I find the problem that the decoding does not work...
Is there a boot argument, fake-id or something else to fix it or do we just have to wait?
I take this opportunity to congratulate you and thank you for your work....
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ChefKissInc, 2023-08-26:
Be patient. We're only 3 students.

aleelmaitro, 2023-08-26:
@ChefKissInc 
I am pleased to report that the decoder seems to be working!!!!!
The modifications I have made are:
VRAM in BIOS: Auto
boot-args: -nredvcn
[image]

aleelmaitro, 2023-08-26:
Well... now when I open Google Chrome the PC restarts...

AxerErnest19, 2023-08-26:
> Well... now when I open Google Chrome the PC restarts...
You have to disable hardware acceleration from Chrome options or from the terminal
https://nootinc.github.io/nred#chrome-chromium-based-browsers-and-apps-like-sublime-text-cause-graphical-artefacts-amongst-other-problems

aleelmaitro, 2023-08-27:
> > Bueno… ahora cuando abro Google Chrome la PC se reinicia… 
> > Hay que desactivar la aceleración por hardware desde las opciones de Chrome o desde la terminal 
> > [https://nootinc.github.io/nred#chrome-chromium-based-browsers-and -aplicaciones-como-texto-sublime-causan-artefactos-gráficos-entre-otros-problemas](https://nootinc.github.io/nred#chrome-chromium-based-browsers-and-apps-like-sublime-text-cause-graphical-artefacts-amongst-other-problems)

Yes... I just applied it and it works...
Thank you very much for the information...

aleelmaitro, 2023-08-27:
> @ChefKissInc ¡¡¡Me complace informarles que el decodificador parece estar funcionando!!!!! Las modificaciones que he realizado son: VRAM en BIOS: Auto boot-args: -nredvcn [image]

Regarding the decoder, Hackintool and VDADecoderCheck say that it is working, but "VideoProc Converter 4K" says that it is not supported
[image]

VisualEhrmanntraut, 2023-08-31:
As the firmware used is correct, and the logic in macOS is correct, and AppleGVA is seemingly almost patched fully, I have no idea what is still up with this. Maybe there is an issue with the rest of the firmware since we don’t replace anything other than VCN.

VisualEhrmanntraut, 2023-09-07:
I have unlocked the conversation for discussion of functionality after new firmware injection. Please don't spam with screenshots or asking for EFIs; I will lock it again.

pokewizardSAM, 2023-09-10:
> can someone with renoir try play a movie on kodi - it uses h264 decoding. by some unknown reason long time decoding crashes the driver on raven

i am pretty new to kodi , installed it just 2 min ago , i idont know if i should have mentoined anything else so please tell me if i left something

results of playng a h.264 encoded file , a h.265 file also works fine 
[image]
[image]
[image]
[image]
[image]

VisualEhrmanntraut, 2023-09-10:
One issue is that it doesn't change the capability of how much resolution is supported. VCN 1 supports smaller resolution than VCN 2

VisualEhrmanntraut, 2023-09-18:
After further analysis the issue is more complicated than just the VA bundle. For instance, the Metal Accelerator Shared User Client which has different platform function indices in X6000 vs X5000. In addition, the "SML" class, which amongst other functions, dictates whether there's SAMU, VCN, VCE, etc. Furthermore, the Video Context class of the accelerator. There might be more I missed in my analysis. Don't know for sure about the dependency of the Accelerator Shared class, but the VA bundle definitely uses it from what I can tell.

aleelmaitro, 2023-09-28:
Hello everyone...
I saw the tag that says "Help Wanted"...
I have no knowledge of programming, but I am willing to carry out tests with your proper instruction...
If this is useful to you, then let me know...

VisualEhrmanntraut, 2023-09-28:
I have a testing device of my own, lol.

aleelmaitro, 2023-10-19:
I want to confirm that I was successful in installing macOS Sonoma 14.0 (23A344) on Ryzentosh with Integrated Graphics
Processor: AMD Ryzen 5 3400G 3.7GHz
Motherboard: ASUS A320M-K
Bootloader: OpenCore 0.9.5
Note: VDA Decoder claims to be failing, but I have Metal 3 support and very smooth 4K video playback...

Hotinosin, 2023-10-22:
I want to confirm that I was successful in installing Ventura 13.5.2 on Ryzentosh with Integrated Graphics
Processor: AMD Ryzen 5 4300G 3.8GHz
Motherboard:  A520
Bootloader: OpenCore 0.9.5
Connect: DP
Note: Hachintool display VDA Decoder claims to be failing, but 4K H.264 works good, very smooth
Thx

VisualEhrmanntraut, 2024-03-13:
Existing code has been removed because it's broken. Part of a bigger end goal called "The Update".

itsyecoder, 2024-04-08:
I am using this on an AMD Ryzen 5800u. I believe this issue is causing crashes on apps such as Firefox, ultimaker cura, and onedrive. Is there a command to put into boot-arg or anyway to make these apps somewhat useful till this is fixed? (plz mark as off topic)

VisualEhrmanntraut, 2024-04-08:
https://docs.github.com/en/issues/tracking-your-work-with-issues/about-issues

VisualEhrmanntraut, 2025-11-16:
with the latest release, the kext side of the implementation is now straight forward. however, the userspace patching is not, due to Lilu's limited "run find replace on shared cache file 1353253463467 times every time it's read".