ChefKissInc/NootedRed #37: Cant make NootedRed wrorking with Ryzen 5 3500U

ueh-oaug, 2023-04-15:
hi, thank you very much cuz u was make a way for AMD Laptop

today i was try install macos on my HP Laptop but i cant make acceleration work ( it just show 7mb )
i was use Xcode build nootedRed but the kext not working. try Artifacts version in your github but it make boot conflict with Lilu.kext (lastest). still not working :(

can you give me a way to make acceleration working on my system please?

My Laptop:

CPU: Ryzen 5 3500U APU Vega 8
MAIN: HP version F.18
RAM: 8GB
SSD: Toshiba

can u spend a little time to check my issue please thank you so much. 

[EFI.zip](https://github.com/NootInc/NootedRed/files/11239007/EFI.zip)

ChefKissInc, 2023-04-15:
@ueh-oaug You’re using WhateverGreen with NootedRed. They conflict

ueh-oaug, 2023-04-15:
@ChefKissInc thank you for reply me. so i need remove WhateverGreen, and use NootedRed.kext in your Github Artifacts Actions right

ChefKissInc, 2023-04-15:
@ueh-oaug Yup

ueh-oaug, 2023-04-15:
@ChefKissInc thank you so much, i will try right now. if nothing wrong i will close this issue. thank you so much. hope u have good day!

ChefKissInc, 2023-04-15:
@ueh-oaug You should aos

ueh-oaug, 2023-04-15:
@ChefKissInc yes sir my hero, im trying now

ueh-oaug, 2023-04-15:
@ChefKissInc i tried but i got stuck at here: 
[image]

its show couldn't alooc class Notered. im using kext in here: https://github.com/NootInc/NootedRed/actions/runs/4700867287

ChefKissInc, 2023-04-15:
Are you sure you're using latest Lilu? It says it's not compatible with it

ueh-oaug, 2023-04-15:
@ChefKissInc im sure, i'm using Lilu 1.6.4 which is latest. ill try download again

ueh-oaug, 2023-04-15:
@ChefKissInc thank you so much!!, i had success reloading version 1.6.4 of lilu.kext. Now my system has acceleration. hope this kext will develop further, wish you success in work and life. Now I'm going to fix the rest.

ChefKissInc, 2023-04-15:
No problem. Enjoy your accelerated system

closed 2023-04-15 by ChefKissInc