ChefKissInc/NootedRed #41: I have Nooted Red by hardware acceleration remains disabled.

ezxpro, 2023-04-30:
I carefully read the readme and the issues on this page. I've built a Hackintosh with the following specs:
Motherboard: Asus Prime B450M-GAMING/BR
RAM: 8GB DDR4 3000MHz
VRAM: Reserved 1GB of RAM
CPU: Ryzen 3 3200G
SSD: Kingston SUV400 SATA II
MacOS Version: Big Sur 11.6.6
SMBIOS: MacPro7,1

In any case, I added the AGPMInjector.kext and the ACPI files from this repo, removed whatever green and added Lilu 1.6.4 and configured them in my config.plist. Yet when I boot into MacOS, I see only 7MB of VRAM, and when I do `kextstat`, AGPMInjector is not among the loaded modules either.

Here's my EFI folder, in case it's necessary.
[EFI.zip](https://github.com/NootInc/NootedRed/files/11361478/EFI.zip)

Allen050329, 2023-04-30:
1. Are you sure you did an OC Snapshot after adding all the files?
2. Which version of macOS are you using?

ChefKissInc, 2023-04-30:
@ezxpro Did you add NootedRed itself

ezxpro, 2023-04-30:
@Allen050329 yes I did
@ChefKissInc I thought the AGPMInjector kext itself was NootedRed. Your comment seems to imply they're different things. In that case, I don't know how to proceed.

 I tried following what was on the readme. I see there are C++ headers and source files, aswell as an Xcode related folder in this repo, but since none of that is mentioned in the Readme, I assumed I just needed the Injector kext and the ACPI files.

ChefKissInc, 2023-04-30:
@ezxpro Those are only companions for better functionality. As this kext is in the Alpha stages, there are no official releases yet, so, you will have to grab a build from our automation, in the GitHub “Actions” tab of the repository. Each “run” has build artefacts at the bottom of the page.
The run you should view should be from the “master” branch; any other branches are for testing purposes only and will not work. Please keep in mind that the kext does not function in recovery (including the macOS installer) or when doing updates; Lilu is to blame for that, we can’t work around it. 
Lastly, these issues are… issues: They’re for you to file issues encountered while using NRed in its intended use conditions and intended usage. All 3 issues you have filed are user issues and are not related to NRed itself; if you continue filing issues that are user errors will force us to block you from the org.

closed 2023-04-30 by ChefKissInc

ezxpro, 2023-04-30:
@ChefKissInc 
Ok, there are a few things interesting things that come out of this.
The first is that I see this project had its initial commit about 2 years ago and that for about a year there has been intense development in it. It's quite a feat considering it's unpaid work. I do software development and work I can only wonder how complex it must be to implement low-level functionality in a closed-source OS like you're doing.

That being said, I think the part about NootedRed's being a separate kext, and that it's releases can be gotten in the actions tab, should absolutely be in the readme.  I am a dev and I would never have even imagine it without asking it here. Maybe it'll annoy you, but I bring this mindset from my work environment, where there's a strong emphasis on documenting stuff and writing how-tos.

Devs can't focus too much on documentation though, even then at least the very most basic information should be somewhere, easy at hand. In the case of NootedRed, the fact it's a separate kext is something you're expecting people to guess or not to know, which will generate bothersome discussions like this, because people will have doubts about it and will ask. 

And they'll ask in the wrong place, because the readme doesn't say anything about how you want the issues page to be used. (I'm suspicious to say that, as you could see earlier, I missed information that was in the readme itself, so please, excuse me), 
I see discussions are enabled in the repo and that you didn't like that I created issues to ask for directions. That should be in the readme too. Most people won't figure these things if they're not in the readme 

One example is the [ExplorerPatcher readme](https://github.com/valinet/ExplorerPatcher/wiki#what-next), which explicitly how they want issues and discussions used in their repo.

Having these basic informations gathered in the readme will hopefully prevent distractions like these.

ChefKissInc, 2023-04-30:
@ezxpro Duly noted.