ChefKissInc/NootedRed #47: Ryzen 3 2200g black Screen issue on SMBIOS MacPro7,1 but other SMBIOS like iMac18,x are OK.

shuhung97, 2023-05-12:
I doubt that, issue may caused by builtin IGPU rename function, conflicted with SMBIOS MacPro7,1 AGDP (former WhereEverRed has no AGDP issue on MacPro7,1).

If possible, can you create a child key like “IGPU Rename” on “com.ChefKissInc.NootedRed” dictionary (Yes / No option) for test? 

Currently, I run on SMBIOS iMac18,1 (GIGA B450M Gaming + Ryzen 3 2200g), some apps like Chrome, HW acceleration setting need to be disabled. Others, basically OK except AppleGVAHEVCEncoder not work.

For AMD GPU, AppleGVAHEVCEncoder.bundle supported system are MacPro7,1 and iMacPro1,1 (see board-id on attached png).[image]

ChefKissInc, 2023-05-12:
Other SMBIOSes than `MacPro7,1`, `MacBookPro16,3`, and `MacBookPro16,4` are not, and will not be, supported by us. For the bundle, I have already applied WhateverGreen's (hwgva-id) board-id patch, so I don't know what you're talking about. Furthermore, do not talk about multiple issues in the same issue, especially if there's an issue open for the topic in question already.

shuhung97, 2023-05-12:
Sorry, seems to be my kext is not up to date. cause black screen, update kext and switch back MacPro7,1 now

closed 2023-05-13 by ChefKissInc