ChefKissInc/NootedRed #48: Chromium-based browsers issue

ruslanshchur, 2023-05-14:
Add any other context about the problem here.
Hi! When I trying to run any Chromium-based browser on macOS 11.7.5 Big Sur with NootedRed, I am getting kernel panic (for me it is not showing, so, it feels like that macOS is hanged up), so, can this serious issue be fixed in future releases of NootedRed? P.S. I used it from master branch

System Information:
macOS 11.7.5
AMD Ryzen 5 3500U
AMD Radeon RX Vega 8

ChefKissInc, 2023-05-14:
@ruslanshchur Fetch the `.gpuRestart` files from `/Library/Logs/DiagnosticReports`

ChefKissInc, 2023-05-14:
(Also, Electron apps work fine on my device, and Electron does use Chromium)

ChefKissInc, 2023-05-14:
@jalavoui I still need the gpu crash logs

ChefKissInc, 2023-05-14:
Duplicate of #13. This issue falls under the same category

closed 2023-05-14 by ChefKissInc