ChefKissInc/NootedRed #50: How do i inject the kext?

vasilisfr, 2023-05-20:
Hi! I am new to hackintosh. As i've figured i need to transfer the aml files to the acpi folder in the efi file. and then edit the config.plist file. Do i need to transfer any files in the Kext directory in the oc folder? Thanks in advance! Hope i get a response

vasilisfr, 2023-05-20:
I have the same processor as you, AMD Ryzen 3500u and I am trying to install macOS with no luck whatsoever, at least for the moment

vasilisfr, 2023-05-20:
Also do i need another aml file in the ACPI folder like SSDT-USBX-etc.aml or only the two aml files provided? Thank you!

ChefKissInc, 2023-05-20:
I think it should be pretty obvious such things are not solvable with a simple AML file
https://docs.github.com/en/actions/managing-workflow-runs/downloading-workflow-artifacts
The workflow is CI. We do not have public releases as the kext is not complete yet. Please note this is not a support forum; refrain from asking questions like these (as you are polluting the bug/feature tracking)

vasilisfr, 2023-05-20:
I am sorry! Im new to this. Is there any way to ask questions for help?

ChefKissInc, 2023-05-20:
@vasilisfr Ask in a support forum, like AMD OS X, or our telegram group (https://t.me/NootInc)

vasilisfr, 2023-05-20:
Thanks a lot! 👍

closed 2023-05-20 by vasilisfr