ChefKissInc/NootedRed #52: Trying to boot on ASUS TUF FX505DT shows nothing on display - Link training failed

STEFnr1, 2023-05-21:
**Describe the bug**
When booted, the display is turned off, even though the OS is running. This may be cause by the fact that in Linux the main display driver is the NVIDIA GTX 1650, which isn't supported by macOS. The error shows this in dmesg: Link training failed.

**To Reproduce**
Steps to reproduce the behavior:
Boot the OS

**Expected behavior**
Display should work, but it doesn't.

**Screenshots**
----

**System Information:**
 - macOS Version: 11.7.6
 - CPU: AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx

**Additional context**
----

ChefKissInc, 2023-05-21:
@STEFnr1 This issue is known internally, but we don’t know the cause yet. 
> This may be cause by the fact that in Linux the main display driver is the NVIDIA GTX 1650
Proof? Have you tried disabling the dGPU to see if that’s actually the case?

STEFnr1, 2023-05-21:
Disabled dGPU, but even with the dGPU enabled it still turns off the display. I know the issue is known, but it's better to have track of issues in one place.

ChefKissInc, 2023-05-21:
@STEFnr1 I meant on linux

STEFnr1, 2023-05-21:
I'm not a pro in Linux. How do I do that? I am booting using the live environment (or however you write it xD)

ChefKissInc, 2023-05-21:
@STEFnr1 I’m not a linux user either…sooooo

ChefKissInc, 2023-05-21:
Also, I think there is an instance of this issue on someone with no dGPU, it might be some kind of edge case not implemented in AMDRadeonX6000Framebuffer (it is essentially a macOS port of Display Core, so comparing the logic is mostly trivial)

ExtremeXT, 2023-05-22:
@STEFnr1 Boolean.

ChefKissInc, 2023-05-22:
You can just put -wegnoegpu in your boot args. But that's not gonna help with the link training issue.

STEFnr1, 2023-05-24:
[image]
Without NRed, the integrated display is detected, so the other display is maybe one emulated by NVIDIA..?

STEFnr1, 2023-05-26:
[image]

STEFnr1, 2023-05-26:
Still no display

ChefKissInc, 2023-05-26:
@jalavoui Like I said in https://github.com/NootInc/NootedRed/issues/52#issuecomment-1556293515, there's someone without a dGPU, and experiencing the same issue. The black screen is because of an error caused while doing the link training, as per the title

ChefKissInc, 2023-05-26:
Related dmesg: http://0x0.st/H-i5.txt
```
ASSERT FAILED : result == DDC_RESULT_SUCESSFULL.
ASSERT LOCATION : void dp_set_panel_mode(struct dc_link *, enum dp_panel_mode) :: /Volumes/Code/p4/GoldenG/OpenGL/GLBuild/AmdRadeonLibraries/src/Latest/core/dal/dc/core/dc_link_dp.c:5062 .
[5:0:0] [DAL][EVENT][link:0] >>> type:"Link Training End", status: FAIL, lane_status: 0.
[5:0:0] [DAL][EVENT][link:0] <<< type:"Link Training End", err: kIOReturnSuccess.
```

Alex-Zhangzhipeng, 2023-06-20:
Now, something strange really happen. I plugged in both video connectors, and the DP interface showed that the secondary screen lit up successfully. After sleeping, I unplugged the HDMI, and after waking up, DP can still be used.
[image]
@ChefKissInc

ChefKissInc, 2023-06-20:
@Alex-Zhangzhipeng Another user on our Telegram group reported this too. It’s good to know it’s consistent at the very least

Alex-Zhangzhipeng, 2023-06-20:
Now I reboot , and the DP is still fine , what a good news!
[image]
@ChefKissInc

ChefKissInc, 2023-06-20:
@Alex-Zhangzhipeng Do a cold boot

Alex-Zhangzhipeng, 2023-06-20:
ok, I will try. @ChefKissInc

ChefKissInc, 2023-06-20:
What if you boot into another OS first?

Alex-Zhangzhipeng, 2023-06-20:
Whether it is re-entering from the BIOS or selecting in the OC interface, the screen will not be displayed. @ChefKissInc

ChefKissInc, 2023-06-20:
@Alex-Zhangzhipeng Sorry, I didn’t get that. Could you rephrase?

Alex-Zhangzhipeng, 2023-06-20:
emmmm , sorry English is not my first language. At first , I change the boot device in the bios to start the windows. Then I use OpenCore to start windows. Both of these, make my computer display a black screen @ChefKissInc

Alex-Zhangzhipeng, 2023-06-20:
Or it's better to use picture to explain it, haha..
[image]
[image]

[image]

ChefKissInc, 2023-06-20:
So Windows has a black screen now?

Alex-Zhangzhipeng, 2023-06-20:
no, window is good , but when I return to Mac , Mac is black.

ChefKissInc, 2023-06-20:
@Alex-Zhangzhipeng Alright. Install the DebugEnhancer kext along with adding `-nreddbg` to your boot args. Then, do the trick that gets macOS to display an image on your internal display. Finally, open the terminal and run the following 2 commands:
`sudo /System/Library/Extensions/AppleGraphicsControl.kext/Contents/MacOS/AGDCDiagnose | curl --progress-bar -Ffile=@- https://0x0.st`
`sudo dmesg | curl --progress-bar -Ffile=@- https://0x0.st`
And send the link outputted by each command over here

Alex-Zhangzhipeng, 2023-06-20:
Ok, i will try these , should i use DP only? or i use DMI or both?@ChefKissInc

Alex-Zhangzhipeng, 2023-06-20:
sorry , maybe I misunderstand it , my pc is not a laptop @ChefKissInc 
[image]

ChefKissInc, 2023-06-20:
@Alex-Zhangzhipeng So it’s just normal DisplayPort with the issue? Then the same thing but with your DisplayPort display working

Alex-Zhangzhipeng, 2023-06-20:
sorry,I cannot understand it , my issue is that my DisplayPort cannot work fine when I change Windows to MacOS(windows is ok). HDMI is perfectly fine, DispalyPort is not. @ChefKissInc

ChefKissInc, 2023-06-29:
@jalavoui what

hhk02, 2023-07-02:
Hello i have the same laptop

hhk02, 2023-07-03:
I installed sucessfully in Monterey with ASUS TUF GAMING FX505DT 

Works after the installation . First i installed the OS without the kext, and in the post installation i downloaded the kext and works.

STEFnr1, 2023-07-15:
> I installed sucessfully in Monterey with ASUS TUF GAMING FX505DT
> 
> Works after the installation . First i installed the OS without the kext, and in the post installation i downloaded the kext and works.

Hello. Do you have the model with Ryzen 7? If yes, how did you make it work?

hhk02, 2023-07-15:
> > I installed sucessfully in Monterey with ASUS TUF GAMING FX505DT
> > Works after the installation . First i installed the OS without the kext, and in the post installation i downloaded the kext and works.
> 
> Hello. Do you have the model with Ryzen 7? If yes, how did you make it work?

First I have installed macOS Monterey without the kext NootedRed once installed I have enabled the kext with the config.plist
Then restart the laptop and enable Secure Boot apply the changes in the BIOS when it reboots, disable Secure Boot again (This will enable the AMD GOP driver since if the Unknown Driver appears it could not continue loading the system) (I think that this thing that I have discovered disables the CSM) Make sure that the SMBIOS is MacBookPro16,3 otherwise the screen will be black.

If everything went well, when the system loads the screen will go black in a few seconds and then the apple will appear as a result the internal graphics will be running.

truongvantuan, 2023-07-15:
> > > I installed sucessfully in Monterey with ASUS TUF GAMING FX505DT
> 
> > > Works after the installation . First i installed the OS without the kext, and in the post installation i downloaded the kext and works.
> 
> > 
> 
> > Hello. Do you have the model with Ryzen 7? If yes, how did you make it work?
> 
> Yes i have Ryzen 7 3570H with AMD Vega 10 too.
> 
> 
> 
> First I have installed macOS Monterey without the kext NootedRed once installed I have enabled the kext with the config.plist
> 
> Then restart the laptop and enable Secure Boot apply the changes in the BIOS when it reboots, disable Secure Boot again (This will enable the AMD GOP driver since if the Unknown Driver appears it could not continue loading the system) (I think that this thing that I have discovered disables the CSM) Make sure that the SMBIOS is MacBookPro16,3 otherwise the screen will be black.
> 
> 
> 
> If everything went well, when the system loads the screen will go black in a few seconds and then the apple will appear as a result the internal graphics will be running.
> 
> 
> 
> I recommend change Framebuffer to 512M to Auto (this increase to 2GB VRAM) for better experience.

VRAM set to Auto, then we can get 2GB vram? Is it possible?

hhk02, 2023-07-15:
I tested and i have 2GB

STEFnr1, 2023-07-15:
> VRAM set to Auto, then we can get 2GB vram? Is it possible?

Can confirm this is true. Setting the setting for the iGPU in BIOS to Auto makes the iGPU have 2GB of VRAM upon boot.

truongvantuan, 2023-07-15:
> I tested and now i have 2GB

I have set Auto but just got 512MB VRAM, thanks bro

hhk02, 2023-07-15:
> > I tested and now i have 2GB
> 
> I have set Auto but just got 512MB VRAM, thanks bro

[image]

VisualEhrmanntraut, 2023-08-31:
Additional information: The macOS kexts seemingly were based on Linux 5.12. However, we had one of the people run Linux 5.0, and that worked fine. We are even more lost than before :/

VisualEhrmanntraut, 2023-10-20:
Caveat: Recovery still gets a black screen. Main OS works fine though.

commit 9e78893b (2023-10-20): Add some delay after dp_receiver_power_ctrl

Fixes #52
closed 2023-10-20 by VisualEhrmanntraut