ChefKissInc/NootedRed #61: ACPI error on AMD ryzen 5 5500u nootedred kext

Berkan111111, 2023-05-25:
ACPI error on AMD ryzen 5 5500u nootedred kext

ChefKissInc, 2023-05-25:
@Berkan111111 Skill issue

Berkan111111, 2023-05-25:
Why?

Berkan111111, 2023-05-25:
Why am I getting this error

Berkan111111, 2023-05-25:
Laptop:Asus vivobook 15 x513ua 
Ram:8GB
[image]
EFI:https://onedrive.live.com/?authkey=%21AKM9s8sFLZPuoLY&id=83E8AF2D3EA2BA57%214076&cid=83E8AF2D3EA2BA57&parId=root&parQt=sharedby&parCid=B75E416A33C5E0B1&o=OneUp

Berkan111111, 2023-05-25:
When Nootedred kext is added is the same error and when missing it is the same error.Can you send me an edited efi

ChefKissInc, 2023-05-25:
If it is not specific to using NootedRed, then it is not an issue that concerns us. Please refer to a support forum like AMD OS X for installation support

closed 2023-05-25 by ChefKissInc