ChefKissInc/NootedRed #65: I have iGPU acceleration, but my screen turns into a green and pink glitch after boot

Hachi-R, 2023-05-29:
after i do all steps to use nootedred kext, i got iGPU acceleration, but my screen glitchs 

my cpu is: Ryzen 5 5600g 
my iGPU is: Radeon Vega 7
i use a HDMI monitor

ruska112, 2023-05-29:
Can you share me your EFI folder?
I'm stuck on a black screen after boot

ChefKissInc, 2023-05-29:
@Hachi-R >HDMI
Yeah uh I don’t think I’ve ever seen a system that uses HDMI but doesn’t have a corrupt display stream

womengzouzailushang, 2023-05-29:
是的，我也是5600G，HDMI输出，屏幕紫色绿色，

Allen050329, 2023-05-29:
Can you guys try using DisplayPort as output, for the sake of science and debugging?

womengzouzailushang, 2023-05-29:
我5600G  HDMI输出也是这样故障（我主板没有DP接口）；我还有个朋友5600G  DP输出是黑屏状态

NyanCatTW1, 2023-05-29:
開 ColorSync, 檢查一下你們顯示器的 Color profile 是 RGB 還是別的
Open the ColorSync Utility and check if your monitor is using the RGB color profile or something else

felicia-wen, 2023-05-30:
HDMI as primary display works fine here, DP as primary results in no signal

womengzouzailushang, 2023-05-30:
系统请升级到Monterey，更新使用原生支持Monterey版本的NootedRed驱动就可以正常显示了，我5600G，HDMI输出紫屏绿屏是这样解决的

Sayor0630, 2023-06-08:
> after i do all steps to use nootedred kext, i got iGPU acceleration, but my screen glitchs
> 
> my cpu is: Ryzen 5 5600g my iGPU is: Radeon Vega 7 i use a HDMI monitor

Can You give me the EFI file you are using? I am also using 5600G. But I can't find the EFI. Please give me your EFI

ChefKissInc, 2023-06-08:
@Sayor0630 No.

Sayor0630, 2023-06-11:
> @Sayor0630 No.

But I can't understand what to do. How to use this Nootedred. Where is the actual file? And where should I put what?

ChefKissInc, 2023-06-11:
Two words. Github Actions

Alex-Zhangzhipeng, 2023-06-19:
My mini PC(5800H)‘s DP output is a black，the Backlight is lighting ，but it dosen't show image.

ChefKissInc, 2023-06-19:
@Alex-Zhangzhipeng That’s #52

Sayor0630, 2023-06-24:
> Two words. Github Actions

What Github Actions. tell me how to get it. Please

ChefKissInc, 2023-06-24:
@Sayor0630 https://docs.github.com/en/actions/managing-workflow-runs/downloading-workflow-artifacts
Filter by branch “master”, for workflow “CI”

Hachi-R, 2023-06-25:
> 

https://github.com/luchina-gabriel/BASE-EFI-AMD-RYZEN-THREADRIPPER

Hachi-R, 2023-06-25:
> > @Sayor0630 No.
> 
> But I can't understand what to do. How to use this Nootedred. Where is the actual file? And where should I put what?

here is a base for ryzen https://github.com/luchina-gabriel/BASE-EFI-AMD-RYZEN-THREADRIPPER , just edit files for your own pc

Hachi-R, 2023-06-25:
> @Hachi-R >HDMI Yeah uh I don’t think I’ve ever seen a system that uses HDMI for the primary display and doesn’t have a corrupt display stream
my mb does't has dp

Sayor0630, 2023-08-15:
> The system please upgrade to Monterey, update the use of native support Monterey version of the NootedRed driver can be displayed normally, I 5600G, HDMI output purple screen green screen is solved like this

Hi. Can I have your EFI file?

VisualEhrmanntraut, 2023-09-07:
Is this issue still relevant?

maff1989, 2023-09-07:
I had a similar issue using a 1440p monitor. The screen would occasionally flicker and go black, then come back on, then repeat. Sometimes it would work properly but it was hit or miss. My 1080p monitor(s) worked fine. I haven't tried with the latest kext from yesterday, I'll report back once I'm able to test again.

BTW I'm running:
AMD Ryzen 7 5825U with Radeon Graphics
Family: 19h, Model: 05h

VisualEhrmanntraut, 2023-09-07:
I meant with the latest kext. And, the author said that it is resolved by updating to Monterey. This is at lowest priority right now as it's a rare issue and specific to BS (Catalina untested)

maff1989, 2023-09-08:
To provide my report, I'm using the latest kext and had my dual 1440p displayport monitors plugged into my Lenovo USB-C docking station, which was routed via USB-C cable into my laptop. Only 1 monitor was detected by macOS, and kept disconnecting and reconnecting without me touching the cables. Unfortunately don't have any additional information to provide besides that, but it seems the issue isn't just related to BigSur as I'm running 13.5.1. Happy to run more tests to provide logs, just not sure where exactly to find those logs.

VisualEhrmanntraut, 2023-09-08:
Your issue is different to the OC; this issue was the screen being corrupted, not constantly connecting and disconnecting. Are you sure your cables aren't bad? And, which SMBIOS are you using

b00t0x, 2023-09-08:
@maff1989 macOS does not support MST, you can only use mirroring even if you connect two displays to the dock.
It's the same on Mac, so I don't think it's a problem of NootedRed or Hackintosh.

VisualEhrmanntraut, 2023-10-04:
Guess it's not relevant anymore.

szmarczak, 2024-03-26:
I can reproduce this by launching League Client:

[image]

The screen freezes for a few seconds but then it returns back to normal.

VisualEhrmanntraut, 2024-03-26:
No. That's #158

Sooly890, 2024-11-29:
With the exact same CPU, HDMI output is completely garbled, and does not work when I set the resolution to 170HZ.

VisualEhrmanntraut, 2025-03-22:
Is this still happening with latest commit?

Sooly890, 2025-11-17:
```
...
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
process efilogin-helper[513] crossed memory high watermark (250 MB); EXC_RESOURCE supressed due to audio playback.
GTrace synchronization point 95
[6:0:0] [AGDC] AMDRadeonX6000_AmdAgdcServices::auxStreamTransfer() ??? AUX Not Supported on This Port: 1.

[6:0:0] [AGDC] AMDRadeonX6000_AmdAgdcServices::auxStreamTransfer() ??? AUX Not Supported on This Port: 1.

[6:0:0] [AGDC] AMDRadeonX6000_AmdAgdcServices::auxStreamTransfer() ??? AUX Not Supported on This Port: 1.

[6:0:0] [AGDC] AMDRadeonX6000_AmdAgdcServices::auxStreamTransfer() ??? AUX Not Supported on This Port: 1.

[6:0:0] [AGDC] AMDRadeonX6000_AmdAgdcServices::auxStreamTransfer() ??? AUX Not Supported on This Port: 1.

GTrace synchronization point c0
[6:0:0] [CUR:0] saveCursorPosition() !!! Cursor Image has not been set.

[6:0:0] [CUR:0] saveCursorPosition() !!! Cursor Image has not been set.

apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
apfs_keybag_init:2037: failed to initialize volume keybag, err = 2
...
```

NootedRed seems not to like HDMI anymore on Big Sur anymore.

NOTE: This is a KVM, but it has successfully booted Ventura, Sequoia, and Tahoe.

Sooly890, 2025-11-17:
Yes, sorry I was not very clear. It is indeed black screening, my monitor (Xiaomi) does not recognise it as even being on.

VisualEhrmanntraut, 2025-11-17:
> Yes, sorry I was not very clear. It is indeed black screening, my monitor (Xiaomi) does not recognise it as even being on.

can you get the debug logs from the latest commit?

Sooly890, 2025-11-19:
> Perhaps your display is reporting 10-bit colour while not actually supporting it?

Well in fact it *does* support HDR however it's currently off (it's a really bad HDR). However, I don't think this is the issue as I've just tested with HDR on and it doesn't like it still.

Sooly890, 2026-06-06:
Same garbled on AOC and black screen on Samsung. Annoyingly serial isn't outputting logs from kexts (still trying to work that one out) but I can provide you with this weird log for now:

[macos_log.txt](https://github.com/user-attachments/files/28660574/macos_log.txt)