ChefKissInc/NootedRed #75: HDMI Black Screen

binxr, 2023-06-04:
Hello again! I use the kext (also very good) but i restart my pc and showing me a black screen, in  HDMI

R5 3350gPRO
VEGA 11
A320MH

ChefKissInc, 2023-06-04:
@binxr

- Big Sur or Monterey (**no** Ventura or Catalina, they are experimental),
- SMBIOS `MacPro7,1`, `iMac20,1` or `MacBookPro16,3`,
- **No** `WhateverGreen`,
- Increased **VRAM Size** to **512MB+**,
etc?

binxr, 2023-06-04:
@ChefKissInc  Monterey 
im using mac pro late 2013. i think it was MacPro6.1
no whatevergreen
2gb vram

ChefKissInc, 2023-06-04:
Change SMBIOS

binxr, 2023-06-04:
that probably will resolve this problem?

ChefKissInc, 2023-06-04:
@binxr It can. No guarantees

binxr, 2023-06-04:
ok ill do it

binxr, 2023-06-04:
@ChefKissInc It worked! thank you so much!

ChefKissInc, 2023-06-04:
That's great. Enjoy!

closed 2023-06-04 by ChefKissInc

ChefKissInc, 2023-06-05:
@lucasolegario English only please. Also, you don’t need to reinstall macOS when changing serials, only to sign out of iCloud beforehand and reset NVRAM before booting after the change

lucasolegario, 2023-06-05:
@ChefKissInc thank you

binxr, 2023-06-08:
> @binxr ao alterar o SMBIOS você reinstalou ou só inicializou pelo pen driva com a EFI atualizada?

i initializate on pendrive bro