ChefKissInc/NootedRed #80: Unstable kext (BIOSTAR A320MH)

binxr, 2023-06-06:
I'm having some problems with my vega 11, the efi is within compliance. and I'm running on mac os monterey, it works, but when it boots, it stays on a black screen, if you want I can send the EFI

BIOSTAR A320MH
AMD Ryzen 5 3350gPRO with vega graphics (vega 11)

binxr, 2023-06-06:
kext works normally only once in a while

ChefKissInc, 2023-06-06:
Which branch?

binxr, 2023-06-06:
if you're talking about kext, the newest

ChefKissInc, 2023-06-06:
@binxr There’s no just “newest”. There’s master, add-ventura-support, add-catalina-support, ventura-dbg, catalina-dbg

binxr, 2023-06-06:
sorry, add catalina support is the correct branch

ChefKissInc, 2023-06-06:
@binxr Why are you using the catalina branch for Monterey

binxr, 2023-06-06:
true, excuse my stupidity

binxr, 2023-06-06:
you can say me the correct one?

ChefKissInc, 2023-06-06:
@binxr master

binxr, 2023-06-06:
kernel panic LMAO

ChefKissInc, 2023-06-06:
@binxr Wat

ChefKissInc, 2023-06-06:
@binxr Show it?

closed 2023-06-06 by binxr