ChefKissInc/NootedRed #81: OLED monitor HDR display splash screen

JonsonWzz, 2023-06-07:
5800h + ALC 256

ChefKissInc, 2023-06-07:
@JonsonWzz ?

ChefKissInc, 2023-06-07:
According to my sources, OLED screens have no brightness controls on macOS even on Intel

ChefKissInc, 2023-06-07:
We're gonna need a video of what you mean with the flashing when HDR is on

ChefKissInc, 2023-06-07:
And it looks like HDR does not work on macOS either, the flickering seems to happen on real Macs as well

closed 2023-06-07 by ChefKissInc

baisong57, 2023-06-07:
I also have the problem, My laptop has an OLED screen, and after loading the corresponding kernel and patch, I found that although I can drag the brightness bar when HDR is turned off, the brightness has not actually changed. When HDR is turned on, not only can automatic brightness be used to adjust it, but manual adjustment can also be used, but the problem is that the screen will keep flashing at this time. When I adjust the screen brightness and want to turn off HDR, the screen brightness will not remain unchanged, but will automatically adjust to the maximum value again.