ChefKissInc/NootedRed #82: STUCK AT: ACPI ERROR: [\_SB_.PCI0.GP18.SATA] - Lenovo ideapad slim 5 pro

ueh-oaug, 2023-06-07:
hi, hope u have good day..

i am extremely happy that you make this kext for AMD laptop. and today, i tried to install it on my laptop.

My laptop specs:

Lenovo ideapad slim 5 pro
CPU: Ryzen 7 5800H - 8 core
RAM: 16GB
SSD: SN750
WIFI: unsupported

i try to install Monterey with your lastest Artifacts kext in Action Part but i cant make my laptop bootable. i stuck at ACPI error: STUCK AT: ACPI ERROR: [\_SB_.PCI0.GP18.SATA] . do you know to fix this error?. please help me. thank you very much. 

[image]

Here is my EFI: 
[EFI.zip](https://github.com/NootInc/NootedRed/files/11678847/EFI.zip)

ChefKissInc, 2023-06-07:
@ueh-oaug

ueh-oaug, 2023-06-07:
@ChefKissInc thank you for reply, althought disable it, i still got this error: 

[image]

ueh-oaug, 2023-06-07:
Bios doesn't have much to set, I set the same as the dortania guide.

ueh-oaug, 2023-06-07:
@ChefKissInc thank you, I will look for public help, it seems that AMD Laptop is still too new, no one can help me :(

closed 2023-06-07 by ueh-oaug

ChefKissInc, 2023-06-07:
@ueh-oaug Our telegram group is https://t.me/NootInc. The issues are for NRed itself