ChefKissInc/NootedRed #84: Ventura不能直接开机

htmambo, 2023-06-08:
**Describe the bug**
系统冷开机的时候，Ventura不能直接使用，必须先从一个老点的系统，比如Montery中正常启动一次再重启进入Ventura才可以使用。

**To Reproduce**
Steps to reproduce the behavior:
1. 冷开机
2. 选择Ventura
3. 进度条跑一半后黑屏，无反应

**Expected behavior**
1. 冷开机
2. 选择Montery，开机正常，在登录界面选择重启或者登录后选择重启
3. 选择Ventura
4. 系统可以正常运行

**System Information:**
 - macOS Version: macOS 13.5 (13.4 正式版一样的情况)
 - CPU: AMD Ryzen 5 5500u

**Additional context**
无

Allen050329, 2023-06-08:
@htmambo Please use English

htmambo, 2023-06-08:
When the system is cold booted, Ventura cannot be used directly. It must be started from an older system, such as Montery, and then restarted to enter Ventura before it can be used.

Allen050329, 2023-06-08:
@htmambo Also, this is a known issue with Ventura and Sonoma

htmambo, 2023-06-08:
Thanks for the reply, I will continue to pay attention and test ventura-dbg and add-ventura-support anytime.

ChefKissInc, 2023-06-08:
This issue has been resolved

closed 2023-06-08 by ChefKissInc