ChefKissInc/NootedRed #89: Unable to drive under Venture

weizhuangzheX, 2023-06-10:
Under the same EFI, I can start normally in Mac OS 12.6.6, and in Mac OS 13.4, I automatically restart the loop after running a short section of code. However, if I do not enable nootedred, I can enter the system's graphics memory normally, which is only 7m

And attach the code that flashed by. I used the screenshot from the video to capture the last frame before restarting

CPU：AMD Ryzen 9 5900HX
OC：0.9.2
[image]

ChefKissInc, 2023-06-10:
You pressed enter too early

weizhuangzheX, 2023-06-10:
[image]

ChefKissInc, 2023-06-10:
Is Lilu up to date

weizhuangzheX, 2023-06-10:
> 
That's right. The latest 1.6.5 and 1.6.6 have both been tried

ChefKissInc, 2023-06-10:
Do you have any other GPUs in the system? Or just the iGPU

weizhuangzheX, 2023-06-10:
> 系统中还有其他GPU吗？或者只是iGPU

And RX6800M, I blocked it through - wegnoegpu

SchmockLord, 2023-06-12:
@weizhuangzheX Hi mate. I have the same laptop as you, the ROG Strip G15 AMD Advantage Edition. Also with the 5900HX +6800M. I also can't get any graphics card enabled. Just in vesa mode. Tried Monterey 12.6 and Ventura 13.3.

I have tried with disabling the 6800M. Also tried to inject different device-ids for the 6800M. But I can see in IOReg that it always pulls DF730000, so the spoofing with e.g. BF73000 is not working.

Can someone please share a fresh copy of nootedred.kext? If I try to compile it with Xcode, I get an error.

[EFI_5900HX-6800M_093.zip](https://github.com/NootInc/NootedRed/files/11719186/EFI_5900HX-6800M_093.zip)

AlphaNecron, 2023-06-12:
Use GitHub Actions.

SchmockLord, 2023-06-12:
@AlphaNecron I have no idea how that works. Not everyone on GitHub knows this stuff and is/wants to become a developer to download a kext. I always downloaded the whole repo, opened the XCode Project, and clicked on the build. But that often gives me errors.

AlphaNecron, 2023-06-12:
https://github.com/NootInc/NootedRed/actions/runs/5230006331
[image]
Click on Artifact.

SchmockLord, 2023-06-12:
@AlphaNecron Thank you :)

With the newest kext I get a reboot very early in the boot progress. As soon as I disable the kext, i goes through Welcome Screen again.

With -wegnoegpu.

Is NootedRed a replacement for Whatevergreen? So should I disable WEG or can it stay beside?

ChefKissInc, 2023-06-12:
@SchmockLord Replacement, also you need to filter by branch master, as there’s other builds that you’re not supposed to use yet

AlphaNecron, 2023-06-12:
Use NRed, install MacOS and then re-enable NRed.