ChefKissInc/NootedRed #97: GPU Clock speed stays at maximum all the time - Renoir

Keylem, 2023-06-16:
With using Renoir on Big Sur (R5 5500U), GPU core speed stays always at 1800Mhz (tho memory speed fluctuates between 1200 and 400 automatically). Is it possible to set a power state for the GPU core if possible? ( my machine constantly consume 10W+ because of the GPU power state excluding AMD Power Gadget [so 25W at idle, 15W CPU 10W+ GPU]) It is possible on linux to change it using a boot argument. Reference to the driver --> https://dri.freedesktop.org/docs/drm/gpu/amdgpu.html at "power_dpm_force_performance_level"

Love the work you do guys, good luck!

ChefKissInc, 2023-06-16:
How are you even observing that
Because the AMD Power Gadget is for the CPU

Keylem, 2023-06-16:
I am using [Stats app](https://github.com/exelban/stats) to monitor CPU / GPU load and remaining battery percentage. I am using AMD Power Gadget to get the current CPU usage in Watts (Usually around 15W) and the total usage with the battery widget of the Stats (it is usually around 25W~ constantly)
You can see the core clock and the memory clock behavior on the following pictures (with the load, memory clock goes up to 1200Mhz):
[image]
[image]

ChefKissInc, 2023-06-16:
incorrect values are bound considering there are 0 apps out that know that working machines with AMD iGPUs even exist

Berkan111111, 2023-06-20:
Can you send me which efi you are using?\

Berkan111111, 2023-06-20:
please post it maybe it will work

andreszerocross, 2023-06-25:
Hi, did you increase VRAM for this notebook??? AFAIK mostly AMD Notebook doesn't have BIOS menu to change VRAM. Only Desktop PC

Berkan111111, 2023-06-25:
512 VRAM

war1644, 2023-07-03:
me too, amd r9 5900hx