ChefKissInc/NootedRed #98: Occasional freeze.

maniacx, 2023-06-18:
Part 2:
After booting Monterey. I get occassional freeze. It freezes for 15/20 seconds and continues to work normally.
This mostly happens when opening app or start the app or navigating to setting 
I attached the dmesg. The error starts at Line 738

```
[ 1722.904732]: virtual void IOAccelEventMachineFast2::checkGPUProgress() - Signaling hardware error on channel 0..\
[ 1722.904738]: void IOAccelEventMachine2::signalHardwareError(eRestartRequest, int32_t): GPURestartSignaled stampIdx=0 type=2 prevType=0 numStamps=35\
[ 1722.904741]: void IOAccelEventMachine2::signalHardwareError(eRestartRequest, int32_t): GPURestartEnqueued stampIdx=0 type=2\
[ 1722.904743]: virtual void IOAccelEventMachineFast2::checkGPUProgress() - Signaling hardware error on channel 2..\
[ 1722.904745]: void IOAccelEventMachine2::signalHardwareError(eRestartRequest, int32_t): GPURestartSignaled stampIdx=2 type=2 prevType=0 numStamps=35\
```

Part 1 :
-Fresh install, first time boot. Booted used nootred to recovery,
-Erased Hardrive
-Reinstalled Monentery. Process started.
-First reboot, installation continues
-Second reboot, installation stalls.
-Reattempted by hard reboot, but still stalls.
-reported as igpu stalled
-Switched to whatevergreen, installation completes
-initial setup of os complete. (such as Language setup,Account info, privacy, etc etc)
-But with whatevergreen device is very slow, no option to scale (everyting is tiny).
-Switched to nootred and UI is fast, screen fluctuates alot due the HDR enabled by default.
-But switching HDR off, screen no longer fluctuates, UI is very responsive, and I get option to scale display.

The stall on happens during installations second boot.
Normally nootedred boots fine.

One thing I noticed in macos system settings it shows VRAM 512MB
I do not have any option to increase the vram in the bios settings
Could this be due to VRAM? Are there any other ways for me to increase VRAM?

**To Reproduce**
Installed Monterey twice. Both times had to switch to whatevergreen to continue installation, that get stalled on second boot.
Freezing occurs frequently.

**Screenshots**

[image]

**System Information:**
 - Laptop: Asus Viwobook M3401QC
 - macOS Version: [e.g. macOS 11.7.4]
 - CPU: AMD Ryzen™ 7 5800H with Radeon™ Graphics × 16
 
  Note: Dedicated Nvidia RTX3050 GPU is Disabled using SSDT-GPU-DISABLE.aml and SSDT-NoHybGfx.aml as per dortania guide.
 - Gpu: NVIDIA GeForce RTX™ 3050 Laptop GPU / AMD Radeon™ Graphics
 - Laptop internal display: OLED, 2880x1800 (16:10), 90Hz
 - Ram: 16GB
 - SMBios: macbookpro16,3

ChefKissInc, 2023-06-18:
at least 1GB VRAM is required for proper operation as per the README. If your BIOS doesn't allow this, use a tool like Smokeless_UMAF

closed 2023-06-18 by ChefKissInc

maniacx, 2023-06-19:
Thanks for the information on Smokeless_UMAF. I have increased the VRAM to 2GB.
However the freezing issue is still there but doesnt happen as often as 500MB.
Just for normal usage such as browsing and using UI, (I dont intend to game), how much VRAM is optimum?

ChefKissInc, 2023-06-19:
What applications are you using when the freezes happen

maniacx, 2023-06-21:
It happens when I am going to settings menu, or check wifi / brightness /bluetooth menu on the panel (quick settings near date/time), or sometime when I click on a icon on dock or desktop. Usually when some graphical activity is taking place. While browsing web or typing on a doc, it doesn't freeze, but if I do something like resizing of windows, or opening settings, panel menus it freezes. 
[image]

I apologize for not attaching the whole dmesg on my first post(I though I did), so I will attach the new one after changing my VRAM to 2GB. Have a look at the dmesg and let me know if it is nootred related (It might not be). As soon as the freeze happened it took the dmesg so the errors are at the end of the file. 

[dmesg.txt](https://github.com/NootInc/NootedRed/files/11811645/dmesg.txt)

maniacx, 2023-06-29:
Just an update.
By trial and error, i found these properties in few of the Ryzen laptop EFI configs and there are **no freezes anymore.**
I have added these in Under UEFI>ReservedMemory section in config.plist. 

		<key>ReservedMemory</key>
		<array>
			<dict>
				<key>Address</key>
				<integer>268435456</integer>
				<key>Comment</key>
				<string>HD3000: IGPU memory corruption errata</string>
				<key>Enabled</key>
				<false/>
				<key>Size</key>
				<integer>268435456</integer>
				<key>Type</key>
				<string>Reserved</string>
			</dict>
			<dict>
				<key>Address</key>
				<integer>569344</integer>
				<key>Comment</key>
				<string>Fix black screen on wake from hibernation for Lenovo Thinkpad T490</string>
				<key>Enabled</key>
				<false/>
				<key>Size</key>
				<integer>4096</integer>
				<key>Type</key>
				<string>RuntimeCode</string>
			</dict>
		</array>

I dont know how or why and I couldnt get any information on these properties