ChefKissInc/NootedRed #99: unstable kext (A320MH-BIOSTAR RYZEN 5 3350GPRO VEGA 11)

binxr, 2023-06-19:
well, I don't know why but there is a black screen with the kext almost every time I start MacOS, but sometimes it works and recognizes my video card, that's very strange.
if the problem is the version of nootedred that I'm using, please tell me the right one.

i'm using the latest version that not marked with NOT USE, (catalina support)

binxr, 2023-06-19:
@ChefKissInc in the case allocateHWEngines wrap is the right one? is marked with master

ChefKissInc, 2023-06-19:
@binxr Yes

binxr, 2023-06-19:
@ChefKissInc if this was my solution i'll cry haha

binxr, 2023-06-19:
@ChefKissInc black screen again after put the right one in EFI

ChefKissInc, 2023-06-19:
@binxr Which macOS version are you using?

binxr, 2023-06-19:
@ChefKissInc monterey man

ChefKissInc, 2023-06-19:
@binxr And the SMBIOS?

binxr, 2023-06-19:
@ChefKissInc MacPro7,1

ChefKissInc, 2023-06-19:
@binxr MacPro7,1 has black screen issues as per the README. If you want to help us debug, we’ll need logs. The process is slightly involved and time consuming. If you don’t want to indulge in it, try iMac20,1

binxr, 2023-06-19:
@ChefKissInc i'll try iMac20,1 first, my gmt-3 says thats 5:57am

binxr, 2023-06-19:
@ChefKissInc bro, i reboot 3 times and right off the bat it worked, if i have any problem i'll back here to explain bye and thank you!

closed 2023-06-19 by binxr