ChefKissInc/NootedRed #100: Can't booting with NootedRed.kext

fahmisrk, 2023-06-20:
hello, i just dived into Hackintosh i can't booting with NootedRed.kext on my AMD Laptop, when i add NootedRed it will get quick restart when try to booting into MacOS

my specs : 
• Processor : AMD Ryzen 3 4300u
• iGPU : Radeon Vega 5
• Storage : WDC PC SN530 NVMe 512GB
• RAM : 2 x 4GB DDR4 3200MHz
• Trackpad I2C HID | Standard PS/2 Keyboard
• Wireless Card : Realtek 8822CE
• Sound Card : Realtek ALC230
• Bootloader OpenCore 0.9.3

what i have tried : 
- boot with essentials kext only
- change npci=0x3 into 0x2
- adding VRAM size into 2GB

[image]

https://github.com/NootInc/NootedRed/assets/61798596/b5b9e150-2105-4225-849b-e718680bcaa4

ChefKissInc, 2023-06-20:
My guess is you used the wrong branch or you have an out of date version of macOS. For the former you need to filter by branch "master"

fahmisrk, 2023-06-20:
I have made sure that I am using the correct branch which is "master". and also I am using macOS Monterey 12.6.6

ChefKissInc, 2023-06-20:
Is Legacy boot/CSM disabled