ChefKissInc/NootedRed #101: Failed allocate memory

ErickFThomaz, 2023-06-21:
**Describe the bug**
I'm using macOs and during use it keeps freezing and crashing

**To Reproduce**
Use system
Watch screen share on discord
Take a screenshot

**Screenshots**
[image]

**System Information:**
 - macOS Version: macOs Ventura 13.4
 - CPU: AMD Ryzen 7 5700G

**Additional context**
[syslog.txt](https://github.com/NootInc/NootedRed/files/11811942/syslog.txt)

ErickFThomaz, 2023-06-21:
Thanks work here!

ErickFThomaz, 2023-06-22:
Before o put in *boot-args* `-nredvcn`command, my chrome, discord and whatsapp there is a long time with a black or white screen before actually opening, I took a look at the logs and I am having the following errors:

If anyone have discord and can help me. DM me *erickfthz*

[discorderror.txt](https://github.com/NootInc/NootedRed/files/11828265/discorderror.txt)
[googleerror.txt](https://github.com/NootInc/NootedRed/files/11828266/googleerror.txt)
[whatsapperror.txt](https://github.com/NootInc/NootedRed/files/11828267/whatsapperror.txt)

killingspree69, 2023-06-22:
@jalavoui what

closed 2023-06-23 by ChefKissInc