ChefKissInc/NootedRed #104: i can't enable nootedred on macos catalina

Tang061022Tang, 2023-06-23:
I installed macOS 10.15.6 on my AMD laptop and set it up successfully without this kext.
But when i enabled this kext, the system couldn't be started and some times later it restarted.
CPU:Ryzen 5 3550H with Vega 8 graphics
macOS version:10.15.6
boot:opencore 0.9.3 with the latest version of NootedRed 1.0.0

NyanCatTW1, 2023-06-23:
From README.md:
> Supports the entire Vega Raven ASIC family (Ryzen 1xxx (Athlon Silver/Gold) to 5xxx and 7x30 series), **from Big Sur to Sonoma.**

closed 2023-06-23 by NyanCatTW1

Tang061022Tang, 2023-06-23:
But I saw it can support it in the latest project
[image]

NyanCatTW1, 2023-06-23:
Mate, it haven't been merged for a reason; so far we've only gotten black screen and failure to start

Tang061022Tang, 2023-06-23:
The same to me😓hope it can support 10.15 in the future