ChefKissInc/NootedRed #106: Enabling nootedred will cause a black screen.

haol24, 2023-06-24:
Hello, my CPU is 5800h, and the system is Mac13.4. After I enable nootedred, the screen will be black. The latest version and previous versions of nootedred have been tried, and mac12 and 11 have also been tried, both of which will be black screen. I will attach a video below.https://github.com/NootInc/NootedRed/assets/133305402/9631611e-72f2-4f8c-a530-a122402f5dab

binxr, 2023-06-24:
@haol24 i think the black screen reason is because your smbios is wrong, which smbios are you using right now?

haol24, 2023-06-25:
> @haol24 i think the black screen reason is because your smbios is wrong, which smbios are you using right now?
> 
> 

The following is my smbios configuration diagram. I don't think this is a smbios configuration problem, because the same model of notebook as me can use nootedred normally, and he hasn't modified smbios.[image][image]

ChefKissInc, 2023-06-25:
We do not support the use of configurators. They often break configs and do not respect kext order. Make your EFI proper

closed 2023-06-25 by ChefKissInc

haol24, 2023-06-25:
> We do not support the use of configurators. They often break configs and do not respect kext order. Make your EFI proper

Maybe you misunderstood. It has nothing to do with efi, because my efi can be used normally in the same model of notebook. I have tried several other efi, but the result is that I can't use nootedred.

ChefKissInc, 2023-06-25:
@haol24 Your SMBIOS is indeed correct, assuming somehow the configurator did not break anything, it is most likely #52

haol24, 2023-06-25:
> @haol24 Your SMBIOS is indeed correct, assuming somehow the configurator did not break anything, it is most likely #52

I have browsed this post. I have tried to add DebugEnhancer kext, but the result is still black.

ChefKissInc, 2023-06-25:
@haol24 The kext is for debugging

haol24, 2023-06-25:
> @haol24 The kext is for debugging

Sorry, I browsed carefully and couldn't find a solution. Should I try to replace smbios?

ChefKissInc, 2023-06-25:
@haol24 There is no solution yet.