ChefKissInc/NootedRed #109: Browser Crashes and Freezes the system (except Safari)

ShioTP, 2023-06-28:
**Describe the bug**
The Browser crashes, and after that, it will freeze the whole system, making it unable to be used.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I hope soon that other browser won't crash and freeze the system. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Version: macOS 13.4.1
 - CPU: AMD Ryzen 5 3400g

**Additional context**
Add any other context about the problem here.

ShioTP, 2023-06-28:
> dup of #91 
Ohh my bad, imma close this then.

closed 2023-06-28 by ShioTP

nanocodium, 2023-06-28:
@ShioTP BOB