ChefKissInc/NootedRed #113: ASUS VIVOBOOK 15 PRO 2022 RYZEN 5 5600H laptop computer OLED screen After turning on the high dynamic range, there is a flash screen.

wzb261643358, 2023-07-04:
Bug description

ASUS VIVOBOOK 15 PRO 2022 RYZEN 5 5600H laptop computer OLED screen After turning on the high dynamic range, there is a flash screen.
Please help me, thank you.

Reproduction Steps:

1、After the system is installed, the high dynamic range is enabled by default in the display driver. After turning it off, the screen no longer flickers, but the screen is relatively dark.
2、After turning on the high dynamic range, there is a flash screen and the brightness can be adjusted. After turning it off, the screen no longer flickers but the brightness cannot be adjusted.
3、I followed the following methods to operate.
"From this repository, add SSDT-PNLF.aml, and SSDT-ALS0.aml if you have no Ambient Light Sensor, 
along with SMCLightSensor.kext for backlight functionality. Usually only works on laptops.
 Add BrightnessKeys.kext for brightness control from the keyboard“

System Information
macOS Version: macOS Ventura 13.4.1

CPU:AMD Ryzen 5 5600H
Screen parameters: 2.8K (2880 x 1620) OLED 16:9 screen ratio, 120Hz refresh rate, 600nit peak brightness, 100% DCI-P3 color gamut, 1,000,000:1 contrast ratio, VESA Certified Display HDR True Black 600 certification.
Video Memory: Set in BIOS to 2GB

wzb261643358, 2023-07-05:
Download the Universal AMD Form Browser file as an EFI and insert it into the UEFI partition of the flash drive. Enter hidden BIOS mode and make changes to the BIOS settings.@SekiroKenjii

SekiroKenjii, 2023-07-05:
@wzb261643358  Thank you so much! It's saved my day <3

baisong57, 2023-07-05:
I have the same problem. When i turn off the HDR, i can not change the screen light.

baisong57, 2023-07-05:
> 

What is the kext? Could you attach a link please? Thank you very much!