ChefKissInc/NootedRed #117: Changing Device Property for VGA Compatible Controller will break Ventura 13.4.1

CraigHazan, 2023-07-08:
This verbose message was a result of changing the VGA compatible controller Device property in the EFI. 

[image]
[image]

ChefKissInc, 2023-07-08:
So what do you mean @CraigHazan

CraigHazan, 2023-07-08:
To change the GPU name for the About This Mac window you could add a device property to rename it in your EFI. A few days ago I had the verbose about failing to boot and I just replaced the EFI and booted to Desktop. Same thing happened again earlier, same verbose and failing to boot. removing the property fixed the problem.

ChefKissInc, 2023-07-08:
Your English quality is making it hard to communicate. If you're using a translator, please switch to a different one.
Do you mean the `model` property?

CraigHazan, 2023-07-08:
I'm from England, there's nothing wrong with my English. Correct the Model number, It says Renoir but about this Mac says AMD Radeon Graphics.

ChefKissInc, 2023-07-08:
Residing in a certain country does not necessarily quantify the extent of your English knowledge.

ChefKissInc, 2023-07-08:
(I still am unable to comprehend what you're trying to say, if you actually want me to resolve this, you will have to find a way to state more clearly your problem)

devio10, 2023-07-08:
> Residing in a certain country does not necessarily quantify the extent of your English knowledge.

[image]

CraigHazan, 2023-07-08:
Nvm

ChefKissInc, 2023-07-08:
Well, if you're trying to override the `model` property, NRed does not support that; it will overwrite it. But I don't know why that'd cause a panic.
But, okay then...

ChefKissInc, 2023-07-08:
Uh... where?

ChefKissInc, 2023-07-08:
@jalavoui Ah, I get it

closed 2023-07-10 by ChefKissInc