ChefKissInc/NootedRed #120: Latest version of NootedRed unable to enter login screen on macOS 12.5

Geekmemz, 2023-07-10:
**Describe the bug** My computer is running macOS 12.5 with an AMD Ryzen 7 5700U CPU. I was previously using an older version of NootedRed (around June 2023) and everything was working fine. However, after updating to the latest version, I encountered a problem. When I enable the -v option and run the code, the boot logo appears on the screen, but it gets stuck there and cannot enter the login screen. **To Reproduce** Steps to reproduce the behavior: 1. Update NootedRed to the latest version 2. Enable the -v option 3. Run the code 4. The boot logo appears on the screen, but it gets stuck there and cannot enter the login screen **Expected behavior** I hope this issue can be resolved so that the latest version of NootedRed can enter the login screen normally.  **System Information:** - macOS Version: macOS 12.5 - CPU: AMD Ryzen 7 5700U **Additional context** My OC version is 0.9.2. Thank you for your attention to this issue. I appreciate your help in resolving it.

ChefKissInc, 2023-07-10:
Does the previous CI run work?

Geekmemz, 2023-07-10:
yes

ChefKissInc, 2023-07-10:
[NootedRed-1.0.0-DEBUG.zip](https://github.com/NootInc/NootedRed/files/12004531/NootedRed-1.0.0-DEBUG.zip)
@Geekmemz does this work?

Geekmemz, 2023-07-11:
It can work, but I panicked when I ran the code with -v for the first time, but the second time I started up smoothly and finished the code. It got stuck on the Apple logo for about 10 seconds, but eventually it went in.

ChefKissInc, 2023-07-11:
But does what I sent work

Geekmemz, 2023-07-11:
I restarted it and it still panicked the first time. After restarting, it finished the code and got stuck on the Apple logo, but this time it has been stuck for 3 minutes and still hasn’t entered the login screen.

Geekmemz, 2023-07-11:
By the way, during a recent accidental entry, I tried running the GPU test of Geekbench 6. The OpenCL test crashed directly, and the Metal test had no progress, but my computer would freeze for a long time.

ChefKissInc, 2023-07-11:
[NootedRed-1.0.0-DEBUG.zip](https://github.com/NootInc/NootedRed/files/12010629/NootedRed-1.0.0-DEBUG.zip)
Does this work?

Geekmemz, 2023-07-11:
It is still stuck on the Apple logo. Do you need me to send you my nootedred that I can use normally?

Geekmemz, 2023-07-11:
Do you need me to send you my opencore EFI？

ChefKissInc, 2023-07-11:
[NootedRed-1.0.0-DEBUG.zip](https://github.com/NootInc/NootedRed/files/12010752/NootedRed-1.0.0-DEBUG.zip)
Can you try this? @Geekmemz

Geekmemz, 2023-07-11:
[image]
This is where it gets stuck after running the code with -v, this is not working anymore.

ChefKissInc, 2023-07-11:
Are you sure you're not using something like `-nredvcn`

Geekmemz, 2023-07-11:
Yes, I used -nredvcn, what is this for and do I need to disable it?

ChefKissInc, 2023-07-11:
Uh... who told you to use it

Geekmemz, 2023-07-11:
I forgot who told me, but I remember someone said that this thing can enable VDA hardware decoding, and turning it on and off does not have much impact.

Geekmemz, 2023-07-11:
So is this option for enabling VDA hardware decoding? If so, why can’t it be turned on?

ChefKissInc, 2023-07-11:
It is, but it's not complete. There is a reason it's gated behind an undocumented boot flag; it's WIP and will cause issues

ChefKissInc, 2023-07-11:
This still shouldn't have happened, and I'm looking into it right now

ChefKissInc, 2023-07-11:
In the meantime, try without the boot arg

Geekmemz, 2023-07-11:
wait，let me try the newest kext

ChefKissInc, 2023-07-11:
Don't. You presented the issue in a way that I thought something else was broken.
Try this one: 
[NootedRed-1.0.0-DEBUG.zip](https://github.com/NootInc/NootedRed/files/12010911/NootedRed-1.0.0-DEBUG.zip)

Geekmemz, 2023-07-11:
What is the difference between the debug version and the release version?

ChefKissInc, 2023-07-11:
[image]

Geekmemz, 2023-07-11:
It works normally, but it has already stuttered in the GPU test of Geekbench 6.

ChefKissInc, 2023-07-11:
Compute not working well is a known and different issue. Thanks for your cooperation in figuring out the issue with `-nredvcn`

commit fb6df044 (2023-07-11): Fix VCN patches

Fixes #120
closed 2023-07-11 by Geekmemz