ChefKissInc/NootedRed #121: Black screen after apple logo

Bee-Hachi, 2023-07-11:
i tryed a lot of noot kext versions (older and newer), mac os montrey and ventura, nvram reset and every time i get stuck on a %@$# black screen

ps. when I switch from noot to whartever green the system boots

CPU: Ryzen 5 5600g 
MB: B450 Asus TUF PRO-S
Connector: HDMI

I'm going completely crazy, I've been trying this for two consecutive days PLEASE SOMEONE HELPE ME

ChefKissInc, 2023-07-11:
@Bee-Hachi SMBIOS?

Bee-Hachi, 2023-07-11:
> @Bee-Hachi SMBIOS?

iMacPro1,1

ChefKissInc, 2023-07-11:
@Bee-Hachi Can you one of the SMBIOSes we recommend in our site?

Bee-Hachi, 2023-07-11:
> @Bee-Hachi Can you try one of the SMBIOSes we recommend in our site?

@ChefKissInc I used this because i saw on the site, which other one you think would work best in my case?

Bee-Hachi, 2023-07-11:
@ChefKissInc MacPro7,1 same erro, but MacBookPro16,3 SUCESS!!!

thank you for your help and thank you so much for your and the team work on this project.