ChefKissInc/NootedRed #122: Freezing and Screen Tearing Issues with Hardware Acceleration Enabled in Edge Browser

Geekmemz, 2023-07-12:
**Describe the bug** After enabling hardware acceleration in the Edge browser, the computer freezes when entering characters in the address bar or playing videos. The freeze manifests as the mouse being able to move, but all other graphical interfaces are not moving. In addition, there is screen tearing when playing videos. When frozen, moving the mouse over the Dock does not enlarge it, even though Dock scaling is enabled in macOS settings. **To Reproduce** Steps to reproduce the behavior: 1. Enable hardware acceleration in the Edge browser 2. Enter characters in the address bar or play videos 3. Observe freezing and screen tearing 4. When frozen, move the mouse over the Dock and observe that it does not enlarge **Expected behavior** I expect the developer to fix this issue as soon as possible. **System Information:** - macOS Version: macOS 12.5 - CPU: AMD Ryzen 7 5700U **Additional context** The startup parameter nredvcn was added and the OC version is 092.
[image]
[image]
[image]

Geekmemz, 2023-07-12:
[image]
[image]
[image]
[image]

KlavierGayming, 2023-07-12:
i most likely am not the one to be replying but it is known that chromium based browsers and some other things dont work properly

Geekmemz, 2023-07-12:
> 

So I submitted this issue, hoping that the developer will solve the problem with the application based on the Chromium kernel as soon as possible.

KlavierGayming, 2023-07-12:
You can disable GPU rasterisation as a workaround for now iirc. Sorry I didn't mention this in my first comment. (or just use safari or firefox)

ChefKissInc, 2023-07-12:
It's in the NRed FAQ

truongvantuan, 2023-07-12:
Could you share us the FAQ link?

ChefKissInc, 2023-07-12:
@truongvantuan It’s literally set as the repo’s homepage and is in the README

Geekmemz, 2023-07-12:
I noticed that this issue has been listed as a known issue, but it has not been resolved yet. This issue is very important to me and I hope it can be resolved as soon as possible. Can you tell me about the progress of resolving this issue? Thank you.

AlphaNecron, 2023-07-12:
Be patient, it's being fixed, not only you waiting for fixes but also everyone does.

ChitoKim, 2023-07-15:
Maybe install a lower version of your browser. In my case chromium downgrade just worked

ChefKissInc, 2023-07-15:
Duplicate of #13

closed 2023-07-15 by ChefKissInc