ChefKissInc/NootedRed #124: [Laptop] Black screen, but everything renders perfectly when using remote desktop

djairhasksha, 2023-07-12:
// I recreated this issue from another GitHub account because my first account is flagged for some reason lmao?

**Describe the bug**
So after installation the screen is blank after boot, but.it seems that graphics work normally.
Before I installed NootedRed I installed AnyDesk and enabled unattended access
After installation & reboot I connected to the laptop, and noticed few things
1. Blur has appeared on Dock & Status bar
2. "Display" property disappeared from "About Mac"
3. macOS recognizes the display as "Built-in Display"
4. Default resolution is not the real screen resolution 
5. Animations are buttery-smooth

**To Reproduce**
Steps to reproduce the behavior:
1. Install NootedRed
2. Restart laptop
3. Enjoy black screen?

**Expected behavior**
A picture on the display

**Screenshots (in my case videos)**
https://youtu.be/Bqcc6SewZt8
https://youtu.be/sHxjY-RAqwM

**System Information:**
 - macOS Version: Ventura 13.4.1
 - CPU: AMD Ryzen 5 4600H
 - Laptop: ASUS TUF Gaming A15 FA506IC

**Additional context**
Without NootedRed the picture is being displayed, but VERY-VERY-VERY laggy and not beautiful 
Backlight is not flashing, it is just disabled, and there is no option to disable HDR (because there's no HDR?)

Geekmemz, 2023-07-12:
Do you have -nredvcn in your startup parameters? If so, try turning it off and trying again.

djairhasksha, 2023-07-12:
> Do you have -nredvcn in your startup parameters? If so, try turning it off and trying again.

Didn't have it in my startup args, rn tried adding this argumant and it made ltrlly 0 diferrence.

ChefKissInc, 2023-07-12:
#52 ?

djairhasksha, 2023-07-12:
> #52 ?

Idk, didn't see anything about link training in dmesg. Btw, i noticed that there are some missing logs before 30 secs of boot when using `sudo dmesg > aaa.txt`. Is there a command to print out full log?

ChefKissInc, 2023-07-12:
@djairhasksha 1. Debugging hooks have been removed from NRed 
2. You need the DebugEnhancer kext, which increases the dmesg size

djairhasksha, 2023-07-13:
Anyway, there's my dmesg. Maybe it will help. Thx for DebugEnhancer hint 
[aaaaaaa.txt](https://github.com/NootInc/NootedRed/files/12038652/aaaaaaa.txt)

ChefKissInc, 2023-07-13:
(?)

djairhasksha, 2023-07-13:
> your card id is not recognized by nootedred
but it still renders everything perfectly?
maybe there's a way to make it forcibly set an id

about errors below - idk how to solve them
is there any advice?

ChefKissInc, 2023-07-14:
These errors are normal

djairhasksha, 2023-07-14:
> check your EFI folder and kexts for old versions / bad installs (google for someone who as a working EFI folder)

I redownloaded every kext and removed useless kexts, and the logs still have these errors.
The joke is: no one has an EFI folder for my laptop. Not even for a close by specs one.

> ps: i assume you have updated to latest ventura

Yes.

djairhasksha, 2023-07-14:
[iiiiii.txt](https://github.com/NootInc/NootedRed/files/12049842/iiiiii.txt)
These are the new logs, if you need then

STEFnr1, 2023-07-15:
Same problems as eDP link failure. #52

ChefKissInc, 2023-07-15:
Duplicate of #52

closed 2023-07-15 by ChefKissInc