ChefKissInc/NootedRed #126: [Laptop] Perpetual boot loop with Renoir (4500U)

eccodev, 2023-07-13:
**Describe the bug**
Upon booting from a USB with EFI and __MACOSX on it, verbose text will scroll for ~10 seconds and then the machine will restart, effectively boot looping.

**To Reproduce**
Steps to reproduce the behavior:
1. Power on laptop
2. Attempt boot from USB
3. Verbose is enabled, so text moves quickly (see video linked below)
4. Boot loop ensues

**Expected behavior**
A successful boot into macOS with working acceleration.

**Screenshots**
I'll do you one better. Here's a video: https://youtu.be/T1U3S6zLbsQ. If you need anything more, let me know and I'll happily provide.

**System Information:**
 - macOS Version: 13.4
 - CPU: Ryzen 5 4500U
 - RAM: 16GB 3200MHz DDR4
 - SSD: Samsung 960 EVO 250GB

**Additional context**
Without NootedRed, I am able to boot into macOS, albeit with no acceleration whatsoever; macOS seems to think it only has 7MB (yes. megabytes) of VRAM. I used Smokeless_UMAF (https://github.com/DavidS95/Smokeless_UMAF) to increase my available framebuffer from 512MB to 4GB. I tried 2GB initially, and moving to 4GB seem to have had no effect on the issue.

joshuafhiggins, 2023-07-14:
Could you upload you're EFI?

eccodev, 2023-07-14:
> Could you upload your EFI?

Sure thing. I hope PixelDrain is fine as a file host. 

https://pixeldrain.com/u/FcEuWnbD

truongvantuan, 2023-07-14:
I did get same issue, clean install macos ventura can resolve this issue.

eccodev, 2023-07-14:
> I did get same issue, clean install macos ventura can resolve this issue.

Good to know. Will try this and update accordingly.

eccodev, 2023-07-15:
Unsure what was changed, but I was eventually able to get NootedRed working. This seems to have been a fringe case or just some weird-ass bug.

closed 2023-07-15 by eccodev