ChefKissInc/NootedRed #127: Will it adapt to the 6800H processor in the future?

8680, 2023-07-14:
Will it adapt to the 6800H processor in the future?

KlavierGayming, 2023-07-14:
Check #92

8680, 2023-07-14:
> 检查#92 （基本上是，但不是优先事项）

Okay, thanks.

closed 2023-07-14 by ChefKissInc

Xryphon, 2023-07-18:
Does anyone know if you can use a supported AMD GPU (i. e. the RX 6650M) without the iGPU enabled? I have a R7 6800H and the RX 6650M but seeing as the iGPU is incompatible but my GPU is, can I still create a Hackintosh?

Xryphon, 2023-07-18:
> @Xryphon If it has a MUX chip, you can use the dGPU.

I don't have one; am I able to disable the iGPU in the plist and use the GPU for setup, or am I stuck?

ChefKissInc, 2023-07-18:
@Xryphon If it can’t redirect the display signal to the dGPU, then what would happen if you disabled it? :-D

Xryphon, 2023-07-18:
@ChefKissInc I guess it could give it a try ;) but alas HP made me believe there was an actual mix switch...