ChefKissInc/NootedRed #128: System crash when running IOS simulator

dungnguyen2703, 2023-07-14:
**System hang then crash then login screen appear when I open IOS simulator**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Open XCode
2. Create new project or open existed IOS project
3. Run project in Simulator
4. Simulator open then system hang and may be crash, then login screen appear

**Expected behavior**
It should work normally

**System Information:**
 - macOS Version: 13.4.0
 - CPU: AMD Ryzen 5 5500U
 - Mini PC Trigkey S5 with Ryzen 5 5500U

ChefKissInc, 2023-07-14:
@dungnguyen2703 VRAM size?

dungnguyen2703, 2023-07-14:
> @dungnguyen2703 VRAM size?

3G

ChefKissInc, 2023-07-15:
Duplicate of #13

closed 2023-07-15 by ChefKissInc

truongvantuan, 2023-07-15:
> > @dungnguyen2703 VRAM size?
> 
> 3G

Hi bro, Did you get  3GB VRAM by Smokeless_UMAF method?