ChefKissInc/NootedRed #131: graphical artefacting on Catalina

RionWren, 2023-07-17:
**Describe the bug**
A clear and concise description of what the bug is. When running macOS Catalina graphics acceleration works but brightness is uncontrollable (it pretends to control but no effect is had) and there is a lot of graphical artefacting, especially in heavier tasks.

**To Reproduce**
Steps to reproduce the behavior:
1. Using latest 'Catalina compatible' version of nootedred
2. install macOS Catalina
3. see bug description4. 

**Expected behavior**
Brightness should be controllable and there should be no artefacting

**Screenshots**
I have attached a recording of the artefacting to this issue: https://github.com/NootInc/NootedRed/assets/139723175/94703248-cd45-4df7-82d5-fc6f63ff06d4

**System Information:**
 - macOS Version: [e.g. macOS 10.15.7]
 - CPU: [e.g. AMD Ryzen 5 4500U]

**Additional context**
Kext works fine on same hardware in Big Sur to Sonoma.

RionWren, 2023-07-17:
adding onto this: the Catalina branch from a quick prelim test has no artefacting but brightness is still uncontrollable

RionWren, 2023-07-17:
graphical artefacting disappears if I have a secondary display connected, disconnecting the display brings back the artefacting and plugging it in while the system is already in macOS hangs the system

ChefKissInc, 2023-07-20:
Brightness will never be controllable in Catalina even if we get all the issues. There's simply no code in it to control the brightness.

RionWren, 2023-07-20:
> Brightness will never be controllable in Catalina even if we get all the issues. There's simply no code in it to control the brightness.

Thanks for the reply! What about the graphical artefacting?

RionWren, 2023-07-20:
> Hi there, any chance I could take a look at your EFI? I can't boot in Catalina, although I suspect it's a processor compatibility. Thanks

No.

ChefKissInc, 2023-07-25:
Please report your issues with experimental code to the respective PR (#72)

closed 2023-07-25 by ChefKissInc