ChefKissInc/NootedRed #135: AGDCC Unauthorized client 'PerfPowerService' AMDRadeonX6000_AmdGpuWrangler Catalina

hhk02, 2023-07-20:
**To Reproduce**
Steps to reproduce the behavior:
1. Install Catalina ( NootedRed works in the installation progress)
2. After finish the install appears the Next line:
AGDCC Unauthorized client 'PerfPowerService' AMDRadeonX6000_AmdGpuWrangler
and stuck there.

**Screenshots**
[image]

**System Information:**
 - macOS Version: Catalina
AMD Ryzen 7 3570H 
AMD Radeon 10 Vega

hi-august, 2023-07-25:
same issue, same efi works on monterey

ChefKissInc, 2023-07-25:
~~How about actually helping us find the issue rather than complaining and making new issues? Do you think we don’t test and know the problems of our experimental branches? Why would we keep the PRs drafted if we thought they were perfectly functional?~~

Duplicate of #134

closed 2023-07-25 by ChefKissInc