ChefKissInc/NootedRed #137: Hangs even when the system is idle (Am I stupid?)

lev666, 2023-07-22:
Good evening. How many times I read telegram or this thread, but I could not be sure until the end. Are the current hangs (as in the video) at this stage of NootedRed acceptable? 

Specification:

Ryzen 7 5800h (vega 8 mobile)
Opencore 0.9.4 release
Latest nootedred (master)

https://nootinc.github.io/nred#constant-freezes-and-crashes-on-ventura-and-newer

Tried also this method of fixing, but no effect was obtained.

I also know there are problems with VdA decoding, but even if the system is idle after opening and closing the "settings" app - the hangs don't disappear at all. They only become intermediate. Will appear for 5-10 seconds and disappear for 1-10 minutes. And so on repeatedly.

https://github.com/NootInc/NootedRed/assets/63856055/ae0d8872-8b81-45c6-b106-8bf38cc670a4

ExtremeXT, 2023-07-23:
@lev666 What macOS version do you have, Monterey? Do you have -nredvcn in boot-args? Fetch the latest file with .gpuRestart extension in /Library/Logs/DiagnosticReports.

lev666, 2023-07-23:
> @lev666 What macOS version do you have, Monterey? Do you have -nredvcn in boot-args? Fetch the latest file with .gpuRestart extension in /Library/Logs/DiagnosticReports.

Thank you for your reply. Yes, I am using Monterey. I do not have this argument in Boot Args. I will upload this file/log a bit later as I am away from my laptop at the moment.

lev666, 2023-07-23:
> @lev666 What macOS version do you have, Monterey? Do you have -nredvcn in boot-args? Fetch the latest file with .gpuRestart extension in /Library/Logs/DiagnosticReports.

Collected all the logs for the .gpuRestart unit.
[all_logs.zip](https://github.com/NootInc/NootedRed/files/12139648/all_logs.zip)

Also rarely, but I see incorrect color transfer (Mouse on this photo has light red edges and paler colors in general). But I read somewhere in Telegramm that problems with color reproduction are still the norm. 

[image]

ExtremeXT, 2023-07-23:
@lev666 Most of the crashes seem to be caused by AvatarPickerMemo, it's a known issue where the compute channels sometimes freeze when doing heavy compute-related tasks. 
The image looks fine to me, I don't see what's wrong.

lev666, 2023-07-23:
> @lev666 Most of the crashes seem to be caused by AvatarPickerMemo, it's a known issue where the compute channels sometimes freeze when doing heavy compute-related tasks. 
> The image looks fine to me, I don't see what's wrong.

The colours are actually distorted, it's just that the camera doesn't show it well

But about this problem "AvatarPickerMemo". I understand that at the moment it is not solvable and I can close this discussion thread?

lev666, 2023-07-25:
> from gpu panic dump i assume montrey
> issue is Compute1 is active while others aren't
> 
> can you try in latest ventura and report ?

I can try to re-install ventura. But when I had it a month ago, it was exactly the same, if not worse sometimes. (that's why I switched to monterey, hoping that the problem is in the newer system).

ChefKissInc, 2023-07-27:
What is your VRAM size

lev666, 2023-07-27:
> What is your VRAM size

2 GB

ChefKissInc, 2023-07-27:
@lev666 Can you try a different one as a test? Like 1GB or 4GB?

lev666, 2023-07-27:
> @lev666 Can you try a different one as a test? Like 1GB or 4GB?

Tried it

I only noticed that the larger I made the size, the less often the problems of hangs appeared even at system idle, but it also did not remove these hangs, but only delayed the time of their appearance.

lev666, 2023-07-27:
>           case 0x1638:
>                 this->chipType = ChipType::GreenSardine;
>                 this->enumRevision = 0xA1;
>                 break;
> 
>         auto targetDeviceId = NRed::callback->chipType >= ChipType::Renoir && NRed::callback->deviceId != 0x1636 ?
>                                   0x1636 :
>                                   NRed::callback->deviceId;
> 
> the driver currently as this workaround 
> try this release if this doesnt work i have no more ideas
> [NootedRed.kext.zip](https://github.com/NootInc/NootedRed/files/12184907/NootedRed.kext.zip)
> 

It's also not a solution

But thanks for the suggestions

lev666, 2023-07-27:
> dam i had a little hope this will work in ventura.
> got an idea if the driver hangs on login screen
> from logs if this line
> Name = AvatarPickerMemo; SubmitContext = Metal,
> means hang on boot screen logo
> the fix is activate auto-login in sys prefs

The freezes appear exactly after the password screen. For example, when I open the settings (after opening the "settings" application, there are a lot of freezes) Or if I just stand idle (without open applications), then rarely, but there are freezes.

ExtremeXT, 2023-07-27:
@jalavoui @lev666 Try reinstalling macOS to make sure there isn't any third party app that's using heavy compute resources. If it works fine on a clean installation, take it slow and install apps one at a time and find the bad one.

lev666, 2023-07-27:
> @lev666 Try reinstalling macOS to make sure there isn't any third party app that's using heavy compute resources. If it works fine on a clean installation, take it slow and install apps one at a time and find the bad one.

Ok, I'll try this a little later, but of all the apps that are currently installed by me are: Tidal, amd sensor/power gadget. But SILENT hangs appear when I open e.g. the system settings themselves. And if I close the settings via "Force close", after a couple of hangs, the hangs themselves become less frequent than with the "Settings" application open. I mean, I don't think it's about the applications I installed

lev666, 2023-07-28:
> you can try disable ryzen cpu kexts + plugins (AMDRyzenCPUPowerManagement.kext, etc)

1)Tried to disable all AMD related cxts except tscsync - it helped. There seems to be no hangs with the "Settings" application open, but also after re-enabling all AMD related cexts - hangs did not appear as they were before

2) With your nootedred cext version in the hacintool app I noticed that VDA Decoder support appeared and also my integrated graphics (with your cext version and the one that is now the latest from nootinc) started to be detected (I didn't pay attention to it before) as Vega 10 Unkown Prototype. But this is probably normal, so maybe some hardware swapping is working etc.And in general I have Vega 8.

ExtremeXT, 2023-07-28:
> bit cosmetics but both NRed kexts kinda the same. i'll bet on disable cpu related kexts for now. you can also try using this flags if it helps [image] if you get hang logs post them here
> 
> this is my kext usage: [image] all fine in ventura

Your no-compat-check argument is incorrectly written, lol. Also disabling SIP won't solve anything.

ChefKissInc, 2023-07-29:
@jalavoui He means that you wrote it incorrectly, so it wasn’t even taking effect; it’s -no_compat_check, not -no-compat-check

lev666, 2023-07-30:
> @lev666 Try reinstalling macOS to make sure there isn't any third party app that's using heavy compute resources. If it works fine on a clean installation, take it slow and install apps one at a time and find the bad one.

For the sake of interest, I decided to install MacOs Sonoma (23A5301g). I did not install any applications. I have observed freezes and severe interface problems only at the stage of system setup. And then there are no hangs, but there are some problems with the display of some content (Example on the photo) 

[image]

[image]

lev666, 2023-07-30:
> sonoma is beta many changes are expected to break Nred. do a fresh ventura install with the kext i send. if all good you will have 0 hangs during installation

I am well aware that sonoma is unstable, both itself and kext to it at the moment. I installed it just for the sake of interest and nothing more

I will try to install ventura now with your kext you sent earlier.

ExtremeXT, 2023-07-30:
> > @lev666 Try reinstalling macOS to make sure there isn't any third party app that's using heavy compute resources. If it works fine on a clean installation, take it slow and install apps one at a time and find the bad one.
> 
> For the sake of interest I decided to install MacOs Sonoma (23A5301g). I did not install any applications. Hangs and strong problems with the interface were observed only at the stage of system setup. And then there are almost no hangs, but there are some problems with the display of some content (Examples on the photo).
> 
> [image]
> 
> [image]
> 
> [image]
> 
> Instead of an orange Avatar, there should have been an animated mouse.

The "Animated mouse" might be what's causing the hangs, try changing the profile picture.

lev666, 2023-07-30:
> @lev666 The "Animated mouse" might be what's causing the hangs, try changing the profile picture.

I tried changing the avatar manually, but that only caused the system to freeze violently for about a minute, and then apparently the video driver would reboot. And it still remained orange.

ExtremeXT, 2023-07-30:
@lev666 Change them from another device then restart your system/.

lev666, 2023-07-30:
> @lev666 Change it from another device then restart your system.

From another device do you mean boot the system from the csd m2s that are in the current laptop on another PC/laptop? If yes, this is not possible as my PC does not support m2 drive initially and so that leaves only my own laptop which is the subject of this discussion.

lev666, 2023-07-31:
> sonoma is beta many changes are expected to break Nred. do a fresh ventura install with the kext i send. if all good you will have 0 hangs during installation

Installed Ventura 13.5 and there are slightly more hangs with the settings app open than in Sonoma, but not so many that you can't use the system at all.

And by the way I noticed that starting with Ventura I stopped having a functioning sleep (on Monterey I only had to fix the ports to make it work). After the laptop goes completely to sleep, 1-3 seconds pass, it wakes up and goes straight into a reboot.

ExtremeXT, 2023-07-31:
> > @lev666 Change it from another device then restart your system.
> 
> From another device do you mean boot the system from the ssd m2s that are in the current laptop on another PC/laptop? If yes, this is not possible as my PC does not support m2 drive initially and so that leaves only my own laptop which is the subject of this discussion. 

@lev666 From mobile phone, website, macOS without NRed, another OS etc. Or just try logging off from your Apple ID. The avatar might be the AvatarPickerMemo that's crashing your driver.

lev666, 2023-07-31:
> @lev666 From mobile phone, website, macOS without NRed, another OS etc. Or just try logging off from your Apple ID. The avatar might be the AvatarPickerMemo that's crashing your driver.

Looks like it's really about the icloud avatar. 

I put a normal photo instead of an animated one - didn't help

Then I logged out of icloud completely and sat for about 15-20 minutes doing various loads (but not particularly demanding) - I didn't see any driver reboot logs or hangs.

As soon as I re-entered icloud - after a minute there was a long freeze and driver reboot log.

Tyningling, 2023-08-05:
I also encountered the same problem. After I tried to downgrade the version from Ventura13.5j to Bigsur11.7.9, this situation improved significantly. Nowadays, I rarely encounter screen freezes. In my Adapted process, I observed some freezing phenomena that are easy to repeat, such as Apple Store search page and screen saver settings

ChefKissInc, 2023-08-07:
I'm pretty sure this (known) issue is also a duplicate of #13

VisualEhrmanntraut, 2023-08-31:
This is likely just a firmware issue as the only firmware we swap right now is the VCN ones, and for the others we just spoof to GC 9.4.0 (which is the Vega 20 dGPU). Likely ditto for the other issue too.

lev666, 2023-08-31:
> This is likely just a firmware issue as the only firmware we swap right now is the VCN ones, and for the others we just spoof to GC 9.4.0 (which is the Vega 20 dGPU). Likely ditto for the other issue too.

And such a small circle of more or less suitable firmware to the embedded vega 8? Which would of course be supported by the macos system. Or is the difficult part the substitution?

VisualEhrmanntraut, 2023-08-31:
Read it again, and again, and again, and again, and again, [...]

VisualEhrmanntraut, 2023-08-31:
I assume all of that is supposed to just ask "Is the difficulty in the compatibility between the "Vega 20" and "the iGPUs" or injecting the firmware". Which still doesn't make sense. 
[image]

lev666, 2023-08-31:
> I assume all of that is supposed to just ask "Is the difficulty in the compatibility between the "Vega 20" and "the iGPUs" or injecting the firmware". Which still doesn't make sense. 
> [image]
> 

That's not what I had in mind a bit, but after seeing the photo I realized my thought was false.

VisualEhrmanntraut, 2023-09-05:
As mentioned in the other issue, I've been working on a fix. It works for all 3 people that have tested so far, including this issue's author. Things like Chrome are still broken though.

VisualEhrmanntraut, 2023-09-05:
5/5 persons now

truongvantuan, 2023-09-06:
> 5/5 persons now

Cool, when you push to github the solution code?

VisualEhrmanntraut, 2023-09-06:
Update: eXT is the only one so far that still has Chrome non-functional

closed 2023-09-06 by VisualEhrmanntraut

NokoPlays, 2024-03-05:
sonoma hangs a lot while doing anything
ventura does not do that at all (from my experience yours maybe different if you use the applications that are currently buggy with nootedred i did not)

commit fb1bb3dd (2023-09-06): Inject all correct firmware. System is now smooth and cool.

Who said macOS on AMD laptops and couldn't work well?
Fixes #137
Fixes #13
Tested-by: ExtremeXT
Tested-by: lev666
Tested-by: Rxxbertx
Tested-by: Yan Schafer
Tested-by: Mert C
Tested-by: Pablom
Tested-by: Truong Tuan
Tested-by: Sapphire
Tested-by: iVintagelover