ChefKissInc/NootedRed #138: Black Screen after apple logo

mzl857, 2023-07-23:
My laptop is Asus TUF Gaming A15（FA506QM)
 CPU : AMD RyzenR7 5800H
Display ：1920x1080 144hz
SMbios：MacBookPro16,3    Intel Core i5-8257U @ 1.40 GHz
OS : Mac os 11
Problem：Black screen after apple logo. I had tried Macos 11-13 , but, it does not work on my device.

ExtremeXT, 2023-07-23:
@mzl857 Does it boot fine without NootedRed? Check if your backlight flickers when on the black screen, if so you need to remote into the laptop and disable HDR. Else, it's a duplicate of #52.

mzl857, 2023-07-23:
> @mzl857 Does it boot fine without NootedRed? Check if your backlight flickers when on the black screen, if so you need to remote into the laptop and disable HDR. Else, it's a duplicate of #52.

without nootedred it boot fine.I will try it tomorrow, thanks for your advice!

mzl857, 2023-07-24:
I had remote into my laptop,nootedred works well,but the overview I can not see my graphics[image]
And The screen is always dark without flickering.

mzl857, 2023-07-24:
I had remote into my laptop,nootedred works well,I can see my video card in the system information.but I can not see my video card in the  overview and my remote tools usually crash.the resolution is not right(not 1920x1080),I tried to change it in the remote tools,but it does not work.The screen is always dark without flickering.So now should I use the solution in #52？
[image]
[image]
[image]

mzl857, 2023-07-24:
> @mzl857 Does it boot fine without NootedRed? Check if your backlight flickers when on the black screen, if so you need to remote into the laptop and disable HDR. Else, it's a duplicate of #52.

I had used dbugenhancer.kext and It sent me back these two links
1.https://0x0.st/HeVq.txt
2.https://0x0.st/HeVT.txt
I don not know what is that mean.
[image]

closed 2023-07-24 by mzl857

ExtremeXT, 2023-07-24:
Duplicate of #52