ChefKissInc/NootedRed #140: Cpu perfor

KemonDream, 2023-07-25:
**Describe the bug**
Black screen

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Version: [e.g. macOS 11.7.4]
 - CPU: [e.g. AMD Ryzen 5 3600G]

**Additional context**
Add any other context about the problem here.

ChefKissInc, 2023-07-25:
Surprise surprise, a chip that includes a GPU and CPU in the same place, performs bad under stress.

KemonDream, 2023-07-26:
Not the gpu performance that I was reporting, it's the cpu mark.
[image]
[image]

ExtremeXT, 2023-07-27:
@KemonDream When your iGPU is also active, the CPU has a little less resources, hence underperforming by a bit. Benchmark scores like this usually mean nothing in the actual usage of the system.

closed 2023-07-27 by ExtremeXT

KemonDream, 2023-07-29:
As I said the percentage is huge and about 10-15%. I don't think a driver should cause that much of resources that I would rather not to ignore.  And also I said I tested it in windows with iGPU driver working the performance marks are in normal stage level. By the way You said the performance mark can't describe actual usage, didi you mean you have tested other application that they were in the same performance in both windows and Mac? Well anyway, thanks for the reply and hardworks on continues developing this driver.

KemonDream, 2023-07-29:
> @KemonDream When your iGPU is also active, the CPU has a little less resources, hence underperforming by a bit. Benchmark scores like this usually mean nothing in the actual usage of the system.
As I said the percentage is huge and about 10-15%. I don't think a driver should cause that much of resources that I would rather not to ignore. And also I said I tested it in windows with iGPU driver working the performance marks are in normal stage level. By the way You said the performance mark can't describe actual usage, didi you mean you have tested other application that they were in the same performance in both windows and Mac? Well anyway, thanks for the reply and hardworks on continues developing this driver.

ChefKissInc, 2023-07-29:
@KemonDream But NRed is not a driver. It’s a patch kext, it’s using Apple’s drivers.

KemonDream, 2023-07-29:
Really interesting that I did a retest. The result is that the power shows in AMD CPU Gadget decreases from 65w(seated in bios PBO) to 50w when disable NRed and enable it. Well maybe it's the apple that did the trick, limiting non-apple's AMD GPU performance. or who knows