ChefKissInc/NootedRed #143: Enable unlimited restart under Ventura

weizhuangzheX, 2023-08-01:
My CPU is r9 5900hx, and using the same EFI, NootedRed can be enabled in Monterey, but it cannot be enabled in Ventura. During Ventura installation, NootedRed was not enabled, and after installation, VESA mode entered the system normally. After enabling NootedRed, the code flashes and then restarts. Subsequently, entering Monterrey again, there was an error report NootedRed nred: @ Unable to obtain VBIOS from VRAM, but I have closed CSM.

cpu r9 5900hx
oc 0.9.3
Lilu1.6.6

panic(cpu 2 caller 0xffffff8020d9905a): NootedRed      nred: @ Failed to get VBIOS from VRAM

Panicked task 0xffffff8b7e3759d8: 61 threads: pid 0: 
Backtrace (CPU 2), panicked thread: 0xffffff9516a39b30, Frame : Return Address
0xffffffebfd3af8b0 : 0xffffff801b47207d 
0xffffffebfd3af900 : 0xffffff801b5c6164 
0xffffffebfd3af940 : 0xffffff801b5b5c77 
0xffffffebfd3af990 : 0xffffff801b412951 
0xffffffebfd3af9b0 : 0xffffff801b47235d 
0xffffffebfd3afaa0 : 0xffffff801b471a07 
0xffffffebfd3afb00 : 0xffffff801bbdb40b 
0xffffffebfd3afbf0 : 0xffffff8020d9905a 
0xffffffebfd3afd80 : 0xffffff801f36dd24 
0xffffffebfd3afe80 : 0xffffff801f3700fe 
0xffffffebfd3afea0 : 0xffffff801b4c81c8 
0xffffffebfd3afee0 : 0xffffff801b4c9258 
0xffffffebfd3affa0 : 0xffffff801b41219e 
      Kernel Extensions in backtrace:
         as.vit9696.Lilu(1.6.6)[7BABC86F-16E1-3542-B898-6136FD272DF8]@0xffffff801f368000->0xffffff801f393fff
         com.ChefKissInc.NootedRed(1.0)[71330C40-01B4-3111-B697-D5B79C194F99]@0xffffff8020d96000->0xffffff8020f77fff
            dependency: as.vit9696.Lilu(1.6.6)[7BABC86F-16E1-3542-B898-6136FD272DF8]@0xffffff801f368000->0xffffff801f393fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1DE0C664-DA38-3436-89ED-1BA79DF0A726]@0xffffff801deac000->0xffffff801deddfff

Process name corresponding to current thread (0xffffff9516a39b30): Unknown
Boot args: alcid=13 -vi2c-force-polling 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 22.6.0: Wed Jul  5 22:21:56 PDT 2023; root:xnu-8796.141.3~6/RELEASE_X86_64
Kernel UUID: 8295C77B-2AAB-3471-81B5-438F59B1A36D
roots installed: 0
KernelCache slide: 0x000000001b000000
KernelCache base:  0xffffff801b200000
Kernel slide:      0x000000001b0dc000
Kernel text base:  0xffffff801b2dc000
__HIB  text base: 0xffffff801b100000
System model name: MacBookPro16,3 (Mac-E7203C0F68AA0004)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization
Hibernation exit count: 0

System uptime in nanoseconds: 449025872
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000000001ac39df7
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000f73f7420d 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8048689000 - 0xffffffa048689000
  . PGZ   : 0xffffff8048689000 - 0xffffff804c68a000
  . VM    : 0xffffff804c68a000 - 0xffffff85189bd000
  . RO    : 0xffffff85189bd000 - 0xffffff86b2023000
  . GEN0  : 0xffffff86b2023000 - 0xffffff8b7e356000
  . GEN1  : 0xffffff8b7e356000 - 0xffffff904a689000
  . GEN2  : 0xffffff904a689000 - 0xffffff95169bc000
  . GEN3  : 0xffffff95169bc000 - 0xffffff99e2cef000
  . DATA  : 0xffffff99e2cef000 - 0xffffffa048689000
  Metadata: 0xffffffc81f4d9000 - 0xffffffc83f4d9000
  Bitmaps : 0xffffffc83f4d9000 - 0xffffffc8454d9000
  Extra   : 0 - 0

lzhoang2601, 2023-08-01:
Disable Legacy Boot, also known as CSM and Boot Mode.

ChefKissInc, 2023-08-01:
https://nootinc.github.io/nred

closed 2023-08-01 by ChefKissInc

weizhuangzheX, 2023-08-01:
I have closed CSM, but it still cannot be started in Ventura. It can be done in Monterey

ChefKissInc, 2023-08-01:
Update macOS Ventura to its latest minor (i.e 13.1 -> 13.5)

weizhuangzheX, 2023-08-01:
I have tried everything from 13.1 to 13.5, and the results are the same

weizhuangzheX, 2023-08-01:
Sorry, it seems that I am unable to upload images here. I hope you can understand the problem and find a solution to it