ChefKissInc/NootedRed #144: Boot failed with NootedRed in Venturare. Removing NootedRed it boots successfully.

Kaijun, 2023-08-03:
**Screenshots**
[image]

[image]

**System Information:**
 - macOS Version: [macOS Ventura](https://www.apple.com/hk/en/macos/ventura/)
 - CPU:  AMD Ryzen 5 5600G
 - MB: MSI Mortar B450m 
 - Connector: HDMI / DP
 - NootedRed Version: [latest](https://github.com/NootInc/NootedRed/commit/a4752d17162f19a1c8796c2b9ff8b41ce418f3f7)
 - Kexts Loaded:
    * AMDRyzenCPUPowerManagement.kext
    * AppleALC.kext
    * AppleMCEReporterDisabler.kext
    * Lilu.kext
    * RealtekRTL8111.kext
    * RestrictEvents.kext
    * SMCAMDProcessor.kext
    * VirtualSMC.kext
    * NootedRed.kext

**Additional context**
I followed the official opencore amd install guide, without NootedRed it boots successfully without any other problem.

but after including NootedRed.kext it crashes and reboots after showing logs in the screenshots.

i'm not pretty sure what's the problem, could anyone take a look. thanks!

btw, here is [my oc config](https://gist.github.com/Kaijun/0c37971a44cd3040a04f82b6d4f15d02)

Kaijun, 2023-08-03:
@ChefKissInc 

sorry i forgot to attach the prerequisites information

- GCN 5 or RDNA AMD dGPU: i'm not sure what kind of type is r5-5600g, but i tried both with `wegnoegpu` or without it, it's still rebooting
- VRAM has already been set to 4GB
- Legacy Boot is also disabled
- WhateverGreen.kext is not included in my kexts
- SMBIOS i tried  iMac20,1 or iMacPro1 or MacPro7,1
- macOS version is Ventura

i'm not sure which config did i miss,  it seems that lot of r5-5600G builds have been succeeded with NootedRed

ChefKissInc, 2023-08-03:
can't help any further because picture is of screen's corner with some numbers and memory addresses instead of the actual `panic: ...` message

Kaijun, 2023-08-03:
> can't help any further because picture is of screen's corner with some numbers and memory addresses instead of the actual `panic: ...` message

ok let me try to grab the boot log and see what caused the panic

ChefKissInc, 2023-08-07:
*crickets*

amaterasushen, 2023-08-10:
I'm having the same problem

ChefKissInc, 2023-08-11:
Timed out waiting for required information

closed 2023-08-11 by ChefKissInc