ChefKissInc/NootedRed #147: Cannot boot Catalina with nootedred enabled

RionWren, 2023-08-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the latest version of nootedred master branch
2. Attempt to boot catalina
3. System panics practically immediately

**Expected behavior**
macOS booting 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Version: [e.g. macOS 10.15.7]
 - CPU: [e.g. AMD Ryzen 5 4500U]

**Additional context**
System boots Monterey fine.

[image]

ChefKissInc, 2023-08-09:
Yes uhhh we didn't do any testing on that version with the HDMI stuff. I am aware the patches don't work there though, I'll fix it soon-ish. For now use an earlier commit

RionWren, 2023-08-09:
Using an earlier commit does work though the same graphical artefacting I mentioned in a previous issue persists (using the add Catalina support build from 2 days ago

ChefKissInc, 2023-08-09:
Yeah, that's what I said; that the HDMI patches don't work in Catalina.

ChefKissInc, 2023-08-09:
[NootedRed-1.0.0-DEBUG.zip](https://github.com/NootInc/NootedRed/files/12300548/NootedRed-1.0.0-DEBUG.zip)
Can you try this? @RionWren

RionWren, 2023-08-09:
> [NootedRed-1.0.0-DEBUG.zip](https://github.com/NootInc/NootedRed/files/12300548/NootedRed-1.0.0-DEBUG.zip) Can you try this? @RionWren

yeah sure, no worries

RionWren, 2023-08-09:
Artefacting stays but it boots fine

closed 2023-08-10 by ChefKissInc

RionWren, 2023-08-10:
adding on, the last thing I mentioned in my previous issue with the artefacting still applies. If I have a monitor plugged in there is no artefacting to either the internal panel or to the display. But if its just the laptop display outputting, it artefacts. It's usable at the very least, thanks for actioning the more pressing issue so quickly!

ChefKissInc, 2023-08-10:
@RionWren Hmmm, sounds like the (e)DP issue where the link training fails, but if you lack a dGPU and plug an HDMI monitor in before boot, that (e)DP display starts working. Very interesting!
You’re utmost welcome, btw.

RionWren, 2023-08-11:
will do once I reinstall (was testing sonoma beta)

RionWren, 2023-08-12:
as requested here's an ioreg with and without the secondary display attached
[Archivio.zip](https://github.com/NootInc/NootedRed/files/12326644/Archivio.zip)
both are in the zip

RionWren, 2023-08-12:
another thing is if I disconnect the display after boot the artefacting comes back

RionWren, 2023-08-12:
and reconnecting it goes away.

RionWren, 2023-08-12:
> try this in ventura with no attached displays [NootedRed.kext.zip](https://github.com/NootInc/NootedRed/files/12327609/NootedRed.kext.zip)

would monterey or catalina be fine? those are the only 2 I have on this system

RionWren, 2023-08-12:
On Catalina it just panics, on monterey I get a black screen

RionWren, 2023-08-12:
I don't have Ventura on this system, sorry. I also don't have the storage to make a temporary partition

commit 1f0b2208 (2023-08-10): Fix HDMI patches on Catalina

Closes #147