ChefKissInc/NootedRed #152: Black screen after boot on Asus TUF FX505DU when using nootedred, not detecting Laptop Display

eaglebh, 2023-08-22:
First of all, congrats for the great work on this, I'm seeing a lot of users reporting success using nootedred kext.
Please, when you have some time, please take a look on my issue. Maybe solving this will help someone else with similar problem.

I'm using the last build.

**Describe the bug**
I used the NootInc Guide, then Dortania's opencore guide, and some other guides/forums on the Internet.

I've installed MacOS Ventura using whatevergreen kext, and it boots fine using it, no acceleration of course.
When I change to nooted red, after the debug messages (I've verbose mode enabled), when it tries to go into GUI mode, the screen turns off/goes black.
I'm using 2GB shared memory to vram, and enabled above 4g decoding, both using UMAF tool.

Steps to reproduce the behavior:
1. Install the system after creating custom opencore EFI, with usb mapping, and boot-args disabling dGPU and using ncpi=0x2000. Secure boot disabled. ACPI mapping was done using Windows SSDTtime, added kext to turn dGPU off as well.
2. Use UMAF tool to enable 4G decoding and set 2GB to VRAM, check on windows that its working (need to keep bios setting for vram as auto, or it will use 512MB otherwise)
3. Check it still boots fine and have GUI using whatevergreen.
4. Disable whatevergreen, enable nootedred, and boot.
5. Now after the boot messages, the screen is black.

**Expected behavior**
We expected to see login screen instead of black screen.

**Screenshots**
Screen with whatevergreen:
[image]

Screen with nooted red:
[image]

**System Information:**
- CPU: AMD Ryzen 7 3750H
- GPU: AMD Radeon Vega 10
- RAM: 16 GB
- Motherboard/Laptop Make and Model: Asus TUF FX505DU
- BIOS revision: 316 (latest)

**Additional context**
Sorry to bother you with this one, but I've tried everything that I could. I've seen similar but not same issues reported here, like issue #52  , #120 and #124 . But with what was said I was not able to make it work. Please, let me know if you need more info, I can run commands using the ssh access or using VNC access, one weird thing is that using VNC when nootedred is enabled is more responsive than using whatevergreen.

`sudo /System/Library/Extensions/AppleGraphicsControl.kext/Contents/MacOS/AGDCDiagnose | curl --progress-bar -Ffile=@- https://0x0.st`
[Output of AGDCDiagnose, above command](https://0x0.st/HLht.txt)

`sudo dmesg | curl --progress-bar -Ffile=@- https://0x0.st`
[Output of dmesg command as above](https://0x0.st/HLhw.txt)

[My plist file](https://gist.github.com/eaglebh/0ba16d711d041b7cf81950cb6d625d4e)

Thanks in advance.

ChefKissInc, 2023-08-22:
Duplicate of #52

ChefKissInc, 2023-08-22:
Creating new issues won’t help us fix it.

ChefKissInc, 2023-08-22:
@eaglebh Issues of software are not specific to people. Besides, the issue is still open.

closed 2023-08-22 by ChefKissInc

eaglebh, 2023-08-22:
Thanks for keeping it on radar. Do you have any tip on what to try, look for?

I've seen some stuff specific to intel igpus, but I'm not sure if nootedred shares some options with it, is it worthy trying tips from this docs: [https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching/connector.html](https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching/connector.html) ?

I've also tried these tips, but ioreg was not able to show any display under the framebuffer:
[https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching/connector.html](https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching/connector.html)

ChefKissInc, 2023-08-22:
@eaglebh Not connector issue. Some missing logic; it fails sending the DPCD command to set the panel mode

eaglebh, 2023-08-22:
Is it better to add info on the #52  instead of here? 
I don't know if it helps but the dmesg and agdcdiagnose outputs when booting using whatevergreen are here:
[dmesg](https://0x0.st/HLCA.txt)
[agdcdiagnose](https://0x0.st/HLCm.txt)

if you give some direction on what to look for, or some material on kext development related to this part,  I can try to jump in and help on coding. It would be an interesting journey for me on my "free" time. I've worked on some Linux kernel very small stuff, still very newbie on the area, but would be happy to enter this new world and try to learn and help.

ChefKissInc, 2023-08-22:
@eaglebh AMDRadeonX6000Framebuffer is the kext. Good luck with that :^)

eeshankeni, 2025-02-03:
Thank you for this thread. Disabling nootedred during install and then enabling it in the userland fixed the issue