ChefKissInc/NootedRed #153: Hardware Acceleration Problem

Zahid1734, 2023-08-23:
**Describe the bug**
While testing for hardware acceleration using the VDADecoderChecker
" An error was returned by the decoder layer. This may happen for example because of bitstream/data errors during a decode operation. This error may also be returned from VDADecoderCreate when hardware decoder resources are available on the system but currently in use by another process. VDADecoderCreate failed. err: -12473 "

**Expected behavior**
Hardware acceleration is fully supported

**System Information:**
 - macOS Version: Ventura 13.5
 - CPU: [e.g. AMD Ryzen 5 5600G]
 MB: Asus prime A520M-k

ChefKissInc, 2023-08-23:
Duplicate of #28

closed 2023-08-23 by ChefKissInc

Kanepan, 2023-08-25:
Can this be fixed? Looking forward to solving this problem, thank you