ChefKissInc/NootedRed #156: How do I increase VRAM size?

Trijal08, 2023-08-31:
I see in the NootedRed website from Wayback Machine that I should increase my VRAM size, HOW DO I DO IT!?

Please and Thank You!

VisualEhrmanntraut, 2023-08-31:
@Trijal08 Bro, our site is still up; it just got moved to “chefkissinc” instead of “nootinc”. Also, Smokeless-UMAF. This isn’t an “issue”.

closed 2023-08-31 by VisualEhrmanntraut

Trijal08, 2023-08-31:
> @Trijal08 Bro, our site is still up; it just got moved to “chefkissinc” instead of “nootinc”. Also, Smokeless-UMAF. This isn’t an “issue”.

What the heo is a Smokeless-UMAF?

cornshiba, 2023-08-31:
A tool to change hidden bios settings for AMD Laptops.

Trijal08, 2023-08-31:
> A tool to change hidden bios settings for AMD Laptops.

What is the default VRAM set in BIOS? Is it max?

cornshiba, 2023-08-31:
What? For me its Auto. Smokeless-UMAF Isn't modding the bios. It is just a bootable tool to change hidden bios settings.

Trijal08, 2023-08-31:
> What? For me its Auto. Smokeless-UMAF Isn't modding the bios. It is just a bootable tool to change hidden bios settings.

OK case closed!