ChefKissInc/NootedRed #159: cannot get past the login screen

AbhishekOjha08, 2023-09-06:
**Describe the bug**
I copied the NootedRed kext and removed whatevergreen. Also cleared nvrm 3 times but i cannot get into the desktop. As soon as I enter my password, it keeps on loading and then crashes and reprompts for login password.

**To Reproduce**
Steps to reproduce the behavior:
1. copy NootedRed kext to the efi and change smbios to iMac20,1  and then tried IMac7,1 

**Expected behavior**
I believe the kext is working because the general mouse movement fluency and transparency is visible. I was able to get into desktop once and it was still loading and then it crashed after a while. And let me make this clear, the system was unusable, dock didn't show up. I changed nothing 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**System Information:**
 - macOS Version: [macOS 11.7.9]
 - CPU: [AMD Ryzen 3 3200G]

 - Motherboard : MSI b450 Tomahawk Max
 - RAM : 2x8GB

**Additional context**
I have used Olarila's Vanilla iso.

VisualEhrmanntraut, 2023-09-06:
We don't support Olarila. Don't use anything from them for your own good.

However, there is indeed a compute hang issue on Raven.

MixedVictor, 2023-09-06:
Same here after downloading the latest version, I had to disable autologin to see what was happening. After trying to log in, it would just freeze and crash the system, I got a [kernel panic log](https://textbin.xyz/?6ba048a08db34415#zD4DQwGN6qkQabPaKf85KcH8k55wD6SgtR5VgMGe8q3) if that will help anyone. Also, I'm using a Ryzen 7 3700U.

VisualEhrmanntraut, 2023-09-06:
Commenting out the MEC firmware injection works around the issue. But that's not a solution.

VisualEhrmanntraut, 2023-09-06:
I look forward to it. ~~Provided it's not just disabling MEC or something~~

VisualEhrmanntraut, 2023-09-06:
@jalavoui All the new firmware is taken straight from Windows.

VisualEhrmanntraut, 2023-09-06:
@jalavoui Eh, don’t worry about it.

commit ba4fd561 (2023-09-06): Fix firmware loading wrong ASIC filename bug

Fixes #159
closed 2023-09-06 by VisualEhrmanntraut